pi-hole/PADD #1: Issues with current Pihole release

evlsmurf, 2017-05-14:
Running current version of your fork with the current version of pihole. Below is image of what issues im seeing.

 
[image]

jpmck, 2017-05-14:
_I've separated these problems into 3 issues (Please see [Chronometer2 says out-of-date when Pihole is up-to-date](https://github.com/jpmck/chronometer2/issues/2) and [Router Information is not shown](https://github.com/jpmck/chronometer2/issues/3) respectively.)_

Hi there!

The CPU temperature in Chronometer2 is pulled from `/sys/class/thermal/thermal_zone0/temp` [here](https://github.com/jpmck/chronometer2/blob/master/chronometer2.sh#L58).

I see that the web admin for Pi-Hole tries to get this information from two separate places:

- `/sys/class/thermal/thermal_zone0/temp`
- `/sys/class/hwmon/hwmon0/temp1_input`

Could you cat both of these files and tell me what you get?

`cat /sys/class/thermal/thermal_zone0/temp`
`cat /sys/class/hwmon/hwmon0/temp1_input`

evlsmurf, 2017-05-14:
`cat /sys/class/thermal/thermal_zone0/temp`  i get `46160`
and `cat /sys/class/hwmon/hwmon0/temp1_input` i get `cat: /sys/class/hwmon/hwmon0/temp1_input: No such file or directory`

jpmck, 2017-05-14:
Hmm... `cat /sys/class/thermal/thermal_zone0/temp` is where I'm getting the CPU temperature...

That number is then converted to either °F, °C or °K based on your Pi-Hole settings, which is pulled from `/etc/pihole/setupVars.conf`.

If run:
`cat/etc/pihole/setupVars.conf`
do you see
`TEMPERATUREUNIT=F`
in the list of variables that's displayed (it could equal C or K)?

evlsmurf, 2017-05-14:
no TEMPERATUREUNIT at all

jpmck, 2017-05-14:
Interesting, I was under the impression that that variable would have been set during the set-up process. Luckily, that should be an easy fix to account for no variable being set.

I would imagine that as a temporary fix, changing and saving the temperature unit in the settings should set that variable for you.
[image]

evlsmurf, 2017-05-14:
yes that also fixes the variable not being there

jpmck, 2017-05-14:
Issue has been fixed by commit [e9cae4f](https://github.com/jpmck/chronometer2/commit/e9cae4f15fcc02aa30edd67aa1108c2cca456158). Thanks @evlsmurf!

closed 2017-05-14 by jpmck