pi-hole/PADD #2: Chronometer2 says out-of-date when Pihole is up-to-date

jpmck, 2017-05-14:
_See [Issues with current Pihole release #1](https://github.com/jpmck/chronometer2/issues/1) reported by [evlsmurf](https://github.com/evlsmurf)_

Running current version of your fork with the current version of pihole on raspberrry pi 3. Below is image of what issues im seeing.

<strike>No cpu temp</strike>
Saying i am out of date 
<strike>Not picking up router info</strike>
 
[image]

jpmck, 2017-05-14:
Thanks for letting me know, I'm in the process of fixing this issue.

evlsmurf, 2017-05-14:
No problem anything else u need me to test let me know thanks

jpmck, 2017-05-18:
Issue has been fixed by commit [c5b57e6](https://github.com/jpmck/chronometer2/commit/c5b57e6dde7888179cf8d6ea54abf2ca07f0e396). Thanks @evlsmurf!

closed 2017-05-18 by jpmck

W1lkins, 2017-05-20:
This is still broken for me, see below:

It seems that when we're checking for the `webVersionLatest` we run `pihole -v -a -l`

On my pi-zero this returns `v3.0.1`, but then when grabbing the current webVersion, we're using `pihole -v -a -c` which (on my machine) returns `v3.0.1a`

Obviously these aren't equal to each other so it's showing as my web version being out of date, but it is not.

If I run `pihole -up`:

```
::: Checking for updates...
::: Pi-hole Core:   up to date
::: FTL:            up to date
::: Web Interface:  up to date
:::
::: Everything is up to date!
```

If I navigate to the web interface it also tells me that my version is up to date.

Any more information needed let me know, thanks.

jpmck, 2017-05-20:
Interesting...

When I run `pihole -v -a -c` to get the current version, I get that `v3.0.1`.
When I run `pihole -v -a -l` to get the latest version, I get that `v3.0.1` as well.

Checking for an update also tells me that everything is up-to-date...
```
pi@raspberrypi:~ $ pihole -up
::: Checking for updates...
::: Pi-hole Core:   up to date
::: FTL:            up to date
::: Web Interface:  up to date
:::
::: Everything is up to date!
```

I do see that version 3.0.1 is tagged as the latest release of the [admin page](https://github.com/pi-hole/AdminLTE/releases)...
[image]

I'm not sure what v3.0.1a is supposed to be... I'm not on the development branch.

W1lkins, 2017-05-20:
Strange, I've hacked in a work around for now so I don't mind but do you see the changes in the below changeset on your machine?

https://github.com/pi-hole/AdminLTE/commit/e204cea85a54ff43deb818c9468c4ab8a3cf20bd#diff-828e0013b8f3bc1bb22b4f57172b019d

This changeset seems to be the 3.0.1a merge into master.

Inside of `/var/www/admin/index.php` and `/var/www/admin/scripts/pi-hole/php/tailLog.php` I have the changes from the above changeset.

What's the `git status` output for you inside of `/var/www/admin`

jpmck, 2017-05-20:
I see... I don't see the changes that were made by that commit. 

Git is also telling me that everything is up-to-date.

```
pi@raspberrypi:/var/www/html/admin $ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```

W1lkins, 2017-05-20:
I've just installed pihole inside of a docker container with the command they specify on their website:

`curl -sSL https://install.pi-hole.net | bash`

Looking inside of /var/www/html/admin I do have the changes from the 3.0.1a changeset

[image]

Like I said, I've hacked in a work around so it's not bothering me! Just something to remember if anyone else has the same issue.