pi-hole/PADD #3: Router Information is not shown

jpmck, 2017-05-14:
_See [Issues with current Pihole release #1](https://github.com/jpmck/chronometer2/issues/1) reported by [evlsmurf](https://github.com/evlsmurf)_

Running current version of your fork with the current version of pihole on raspberrry pi 3. Below is image of what issues im seeing.

<strike>No cpu temp</strike>
<strike>Saying i am out of date</strike>
Not picking up router info
 
[image]

jpmck, 2017-05-14:
@evlsmurf, the DHCP server is based on your Pi-Hole settings, which is pulled from `/etc/pihole/setupVars.conf`. Your IPv4/6 address and whether or not the P-Hole is acting as a DHCP server is pulled from here as well, so it appears that the script is able to pull information from it...

If run:
`cat/etc/pihole/setupVars.conf`
do you see
`DHCP_ROUTER=192.168.1.1`
in the list of variables that's displayed (it should be the IP of your router if that's acting as your DHCP server)?

evlsmurf, 2017-05-14:
it was not in there added it and now it works

jpmck, 2017-05-16:
Issue has been fixed by commit [632ac97](https://github.com/jpmck/chronometer2/commit/632ac970c3665a0d975d9957355c55539da2dbd2). Thanks @evlsmurf!

closed 2017-05-16 by jpmck