pi-hole/PADD #4: DHCP Alignment

evlsmurf, 2017-05-15:
any chance you can fix the alignment issues with the DHCP line along with ( )  i tried to mimic the screenshot you have on the main page but i couldn't figure it out 

[image]

jpmck, 2017-05-15:
This was changed in commit [45c43fe](https://github.com/jpmck/chronometer2/commit/45c43fec913ac59ad6b94f1362187594ba6b6922) due to "long" IP address (i.e. 192.168.100.100) ranges wrapping on the PiTFT display.

Example:
<pre>NETWORK ====================================================
 Gateway:  192.168.100.2
 IPv4:     192.168.100.3/24
 IPv6:     2601:542:101:3c6f:c3aa:ee5f:2930:bb6c
 DHCP:     Enabled            192.168.100.100 - 192.168.100.1
50
SYSTEM =====================================================</pre>

I plan on reformatting the display in the future, but don't plan on changing it right now.

closed 2017-05-15 by jpmck

jpmck, 2017-05-20:
This issue has been fixed by commit [b6998c6](https://github.com/jpmck/chronometer2/commit/b6998c63e141f6b793a681296443f1168008ed44).

The DHCP section of the display has been tweaked to either show the IP range being handed out (if Pi-hole is acting as the DHCP server), or the IP address of the DHCP server on the network.

<pre>DHCP SERVER ================================================
 Status:   Enabled            IPv6:     Enabled
 Range:    192.168.100.100 - 192.168.100.150</pre>

or 

<pre>DHCP SERVER ================================================
 Status:   Disabled           IPv6:     N/A
 Server:   192.168.100.2</pre>