pi-hole/PADD #5: Get's printf errors when running script

yeold, 2017-05-19:
./chronometer2.sh: line 67: printf: 55.1480: invalid number
./chronometer2.sh: line 97: printf: 64.00: invalid number
./chronometer2.sh: line 97: printf: 35.00: invalid number
./chronometer2.sh: line 97: printf: 24.00: invalid number
./chronometer2.sh: line 82: printf: 64.0000: invalid number

And then the stats prints out

jpmck, 2017-05-19:
Thanks for letting me know, @yeold!

Do the above errors display on each refresh or the first time that chronometer2 runs?

Could you provide the full contents of the screen (copy/paste) when the error occurs?

yeold, 2017-05-19:
Hi!This is what the printout looks like./chronometer2.sh: rad 67: printf: 55.1480: ogiltigt tal./chronometer2.sh: rad 97: printf: 26.00: ogiltigt tal./chronometer2.sh: rad 97: printf: 24.00: ogiltigt tal./chronometer2.sh: rad 97: printf: 19.00: ogiltigt tal./chronometer2.sh: rad 82: printf: 26.0000: ogiltigt tal ___ _     _        _         Pi-hole is up-to-date!| _ (_)___| |_  ___| |___     Pi-hole v3.0.1|  _/ |___| ' \/ _ \ / -_)    Pi-hole Web-Admin v3.0.1a|_| |_|   |_||_\___/_\___|    FTL v2.7.3PI-HOLE ==================================================== Status:   Active             Blocking: 106239 Queries:  11624              Pi-holed: 1557 (13.39%) DNS 1:    8.8.8.8            DNS 2:    8.8.4.4NETWORK ==================================================== Gateway:  192.168.1.1 IPv4:     192.168.1.4 IPv6: DHCP:     Disabled (Router: 192.168.1.1)SYSTEM ===================================================== Hostname: pi-hole Uptime:   13 days, 5 hours, 42 minutes CPU Temp: 0,0°C              CPU Load: 0.26, 0.24, 0.19 Memory:   13.7%              CPU Load: 0,0%              Press ^C to quit chronometer^Cpi@pi-hole:~ $Some things is in Swedish but I think you can understand what it means.The errors  repeats on every loop.I am running this on a Pi rev B the oldest version if that matters

jpmck, 2017-05-19:
I don't think that you were able to see my edit previously.

I believe that this is an issue with the locale settings on your system...

Could you run `locale` as well as `locale -va` and let me know what is returned?

jpmck, 2017-05-20:
I believe that this issue has been fixed by commit [b6998c6](https://github.com/jpmck/chronometer2/commit/b6998c63e141f6b793a681296443f1168008ed44). Updating to the latest chronometer2, should fix this. If there are any issues with this fix, please let me know. Thanks @yeold !

closed 2017-05-20 by jpmck

Hrothgah, 2018-07-10:
Hi, having the same issue at the moment. 
Locale output: `LANG=it_IT.UTF-8
LANGUAGE=it_IT.UTF-8
LC_CTYPE="it_IT.UTF-8"
LC_NUMERIC="it_IT.UTF-8"
LC_TIME="it_IT.UTF-8"
LC_COLLATE="it_IT.UTF-8"
LC_MONETARY="it_IT.UTF-8"
LC_MESSAGES="it_IT.UTF-8"
LC_PAPER="it_IT.UTF-8"
LC_NAME="it_IT.UTF-8"
LC_ADDRESS="it_IT.UTF-8"
LC_TELEPHONE="it_IT.UTF-8"
LC_MEASUREMENT="it_IT.UTF-8"
LC_IDENTIFICATION="it_IT.UTF-8"
LC_ALL=it_IT.UTF-8`

locale -va output:
`locale: it_IT.utf8      archive: /usr/lib/locale/locale-archive
-------------------------------------------------------------------------------
    title | Italian locale for Italy
   source | RAP
  address | Sankt Jørgens Alle 8, DK-1615 København V, Danmark
    email | bug-glibc-locales@gnu.org
 language | Italian
territory | Italy
 revision | 1.0
     date | 2000-06-29
  codeset | UTF-8

locale: C.UTF-8         directory: /usr/lib/locale/C.UTF-8
-------------------------------------------------------------------------------
    title | C locale
    email | aurel32@debian.org
 language | C
 revision | 1.6
     date | 2016-08-08
  codeset | UTF-8`

and this is what I see when I run padd:
` __      _   __
|__) /\ |  \|  \  Pi-hole® Ad Detection Display
|   /--\|__/|__/  A client for Pi-hole

START UP ===================================================
- Writing PID (5690) to file...
- Checking for PADD version file...
  - PADD version file found... deleting.
- Gathering system information...
./padd.sh: riga 185: printf: 48.3120: numero non valido
./padd.sh: riga 639: printf: 1.00: numero non valido
./padd.sh: riga 639: printf: 2.00: numero non valido
./padd.sh: riga 195: printf: .2500: numero non valido
./padd.sh: riga 636: printf: 14.5: numero non valido
(standard_in) 1: syntax error
(standard_in) 1: syntax error
./padd.sh: riga 663: printf: 1.40: numero non valido
- Gathering Pi-hole information...
./padd.sh: riga 142: printf: 0.751880: numero non valido
(standard_in) 1: syntax error
(standard_in) 1: syntax error
- Gathering network information...
- Gathering version information...
  - Pi-hole Core v3.3.1
  - Web Admin v3.3
  - FTL v3.0
  - PADD v2.1.0
- Checking for update information...
  - Pi-hole is up-to-date!`

("Numero non valido" is italian for "invalid number")

Thank you