pi-hole/PADD #6: Chronometer2 says out-of-date when Pihole is up-to-date

evlsmurf, 2017-05-21:
Running newest version of Chronometer2 was working fine for most of the day saying correct date then changed to out-of-date when Pihole is up-to-date.

also how does one fill in domain section 

[image]

Tehgeekguy, 2017-06-03:
Same issue here, even though pihole -up shows everything is up to do.

evlsmurf, 2017-06-04:
@Tehgeekguy  i deleted my piHoleVersion let it recreate it and i still have the same issue as screenshot above telling me web portion is out of date, did you do any edit to chronometer or just deleted file to recreate.

Tehgeekguy, 2017-06-04:
@evlsmurf no other edits, I just closed the chronometer, removed piHoleVersion, launched chronometer again and when it launched the first time the first line that shows if it's up to date or out of date was missing, so I closed it once more and re-launched and it showed everything was up to date.

evlsmurf, 2017-06-10:
Issue is still present on today’s release of chronometer2 1.1.0

jpmck, 2017-06-10:
It looks like there was a version 3.0.1a released on May 20th, however 3.0.1 is indicated as the latest release:

[image]

However, I'm unable to actually get this version of Pi-hole... A clean install gives me v3.0.1.

Additionally, checking my Pi-hole's version and trying to update it yield's the same result:

```
pi@raspberrypi:~ $ pihole -v
::: Pi-hole version is v3.0.1 (Latest version is v3.0.1)
::: Web-Admin version is v3.0.1 (Latest version is v3.0.1)
pi@raspberrypi:~ $ pihole -up
::: Checking for updates...
::: Pi-hole Core:   up to date
::: FTL:            up to date
::: Web Interface:  up to date
:::
::: Everything is up to date!
```

At this point in time the testing for whether or not the core/web admin/FTL version is up-to-date is a simple check to see if the current version on your Pi-hole is equal to the most recent version on the Pi-hole project's Github. If the two aren't equal, it's shows that your Pi-hole needs updating.

What is the output of the following on your Pi-holes?
```
pihole -v
```

```
pihole-FTL version
```

evlsmurf, 2017-06-10:
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jun 10 11:26:15 2017 from 10.8.0.2

 ___ _     _        _
| _ (_)___| |_  ___| |___     Pi-hole is out-of-date!
|  _/ |___| ' \/ _ \ / -_)    Pi-hole Core v3.0.1
|_| |_|   |_||_\___/_\___|    (Web v3.0.1a, FTL v2.8)

PI-HOLE ====================================================
 Status:   Active             FTL:      Running            
STATS ======================================================
 Blocking: 816851
 Queries:  14773              Pi-holed: 355 (2%)           
NETWORK ====================================================
 Hostname: PiHoleVPN          Domain:                      
 IPv4:     192.168.1.23/24    
 IPv6:                        
DHCP SERVER ================================================
 Status:   Disabled           IPv6:     N/A                
 Router:   192.168.1.1
SYSTEM =====================================================
 Uptime:   0 days, 2 hours, 20 minutes
 CPU Temp: 47.2°C             CPU Load: 0.23, 0.14, 0.10
 Memory:   10.6%              CPU Load: 5.8%               

pi

 ___ _     _        _
| _ (_)___| |_  ___| |___     Pi-hole is out-of-date!
|  _/ |___| ' \/ _ \ / -_)    Pi-hole Core v3.0.1
|_| |_|   |_||_\___/_\___|    (Web v3.0.1a, FTL v2.8)

PI-HOLE ====================================================
 Status:   Active             FTL:      Running            
STATS ======================================================
 Blocking: 816851
 Queries:  14784              Pi-holed: 355 (2%)           
NETWORK ====================================================
 Hostname: PiHoleVPN          Domain:                      
 IPv4:     192.168.1.23/24    
 IPv6:                        
DHCP SERVER ================================================
 Status:   Disabled           IPv6:     N/A                
 Router:   192.168.1.1
SYSTEM =====================================================
 Uptime:   0 days, 2 hours, 20 minutes
 CPU Temp: 47.8°C             CPU Load: 0.23, 0.14, 0.10
 Memory:   10.6%              CPU Load: 5.8%                                         ^C
pi@PiHoleVPN:~ $ 
pi@PiHoleVPN:~ $ 
pi@PiHoleVPN:~ $ 
pi@PiHoleVPN:~ $ clear

pi@PiHoleVPN:~ $ pihole -v
::: Pi-hole version is v3.0.1 (Latest version is v3.0.1)
::: Web-Admin version is v3.0.1a (Latest version is v3.0.1)

pihole-FTL version  2.8

djl236, 2017-06-17:
I updated to the newest Chronometer2 and I'm still having this issue.

Pihole and Web-Admin are both v3.0.1 (latest) and FTL is 2.9 (latest) but its saying my FTL is out of date.

Tried deleting piHoleVersion and rebooting multiple times. Still saying out of date.

jpmck, 2017-06-18:
Chronometer2 v1.2.0, which was just released today should take care of this problem. If there are any issues, let me know and I'll reopen this issue.

closed 2017-06-18 by jpmck

djl236, 2017-06-20:
Its working great now, thanks!

Zalash85, 2018-06-09:
@jpmck I have version 1.3.1 installed and I'm intermittently getting this issue.