pi-hole/PADD #7: Stuck at startup - version file not found

r0ckarong, 2017-06-24:


jpmck, 2017-06-25:
Hi, the version file not existing is okay for the first start up.

Could you tell me what version of Pi-hole you are using?

You can either use ```pihole -v``` from the terminal, or find the version numbers at the corner to the web admin console.

[image]

r0ckarong, 2017-06-26:
Latest official release just like you posted.

```
  Pi-hole version is v3.1 (Latest: v3.1)
  AdminLTE version is v3.1 (Latest: v3.1)
  FTL version is v2.9.4 (Latest: v2.9.4)
```

r0ckarong, 2017-06-26:
What I haven't mentioned is that I don't run an adafruit display but something very similar. I can't even get to setting that up though because it already hangs when I try to start it from an SSH session in a regular terminal emulator.

jpmck, 2018-05-13:
I have been unable to replicate this. Closing.

closed 2018-05-13 by jpmck