pi-hole/PADD #11: Error in 1.2.2 (Permission Denied)

m0n5t4H, 2018-05-13:
when trying to boot by adding chronometer2.sh bash
it refuses to launch and gives and error; Permission Denied

Reverted back to 1.2.1 and the script launches.

jpmck, 2018-05-19:
Could you tell me what you get from a `ls -l` in the directory that you placed chronometer2 in?

m0n5t4H, 2018-05-20:
Updated to 1.3 issue was solved and the script loaded

I had to edit out the LC_ALL
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

any chance we can get the option of either the original logo or the updated one?

jpmck, 2018-05-20:
Closing as the issue can't be duplicated.

The issue with the locale setting is being worked on and tracked with [Issue 13](https://github.com/jpmck/chronometer2/issues/13)

closed 2018-05-20 by jpmck