pi-hole/PADD #16: Multiple 'cannot change locale' warnings

matt-clegg, 2018-06-12:
When I run chronometer2, I get multiple warnings on startup saying:

`/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)`

Which you can see in this [image](https://i.imgur.com/xvx8pIU.png).

 Also, when the chronometer2 is running, I see the same warning at the [bottom of the terminal](https://i.imgur.com/SM8igJL.png). 

I'm running Pi-Hole v3.3.1, Web v3.3, FTL v3.0, Chronometer2 v1.3.1. The chronometer2.sh script is located in the home directory, was made executable and was run as `sudo ./chronometer2.sh`.

jpmck, 2018-06-23:
Since the default Raspberry Pi image is in GB English, I've tweaked the above.

Corrected with [ed23cda](https://github.com/jpmck/PADD/commit/ed23cdafbd2981dfaf407581dfd0038e61a4251d#diff-f0d2b00f20028b2c7f9febd84b9659afR11)

closed 2018-06-23 by jpmck

OfficialMuffin, 2018-09-18:
This is still a problem for me. What can I do?