pi-hole/PADD #17: Using beta version of pihole can mess up display

bpennypacker, 2018-06-14:
I just decided to install the beta version of pihole on one my pi's so that I could play around with the new FTLDNS they're working on. It's resulting in a version header like this:

> Pi-hole v3.3-181-77680, Web v3.2.1-195-43552, FTL v-50

Those long version strings with the build numbers in them cause the chronometer2 display to wrap and throw everything off a bit. Here's what this beta version shows when you run `pihole -v`:

```
$ pihole -v
  Pi-hole version is v3.3-181-ga7e7680 (Latest: v3.3.1)
  AdminLTE version is v3.2.1-195-g4355bde2 (Latest: v3.3)
  FTL version is vDev-5ecab0a (Latest: v3.0)
```

I realize that this is a beta version of pi-hole so I completely understand if you don't want to do anything about it. I was just thinking it'd be nice if chronometer2 would truncate the versions so that rather than `v3.3-181-77680` it just used `v3.3`. Perhaps only do this if a a command line option is specified when running chronometer2. Maybe also change the `Pi-hole is out-of-date!` message to something along the lines of `Pi-hole is running a test build` when the command line option is used as well since the beta version technically isn't out-of-date. 

Or perhaps an entirely new banner at the top when you tell chronometer2 that you're running a test build. Something like this:
```
.-..   .      .      Pi-hole test v3.3-181-77680
|-'. - |-. .-.| .-,  Web test v3.2.1-195-43552, FTL v-50
'  '   ' '-`-''-`'-  Chronometer2 v1.3.1
```

evlsmurf, 2018-06-21:
Yes please do this

jpmck, 2018-06-23:
This will not be implemented at this time.

(Please see [Issue #15](https://github.com/jpmck/PADD/issues/15#issuecomment-390474893).)

closed 2018-06-23 by jpmck