pi-hole/PADD #21: Display scaling for 800x480 displays

chvvkumar, 2018-07-07:
**Is your feature request related to a problem? Please describe.**
PADD scaling is small when using on 5" 800x400 TFT displays

**Describe the solution you'd like**
Is there a parameter which controls scaling (for example, to scale the UI by x%)?

**Describe alternatives you've considered**
Tried "_sudo dpkg-reconfigure console-setup_" but it scales the whole terminal, not just PADD.

**Additional context**
Photo of issue:

[image]

jpmck, 2018-07-08:
Hi, I took a look into this, and …

Is there any easy way for me to fix this within PADD? No, not really.

Is there a decent solution that might work for you? Yes, but you’ll have to do some tweaks to get things looking how you’d like them…

My recommendation (I think this will work…) would be to add in a couple of lines in your `.bashrc` file to change the font of the terminal with `setfont` before and after PADD runs like so:

```bash
# Run PADD
# If we're on the PiTFT screen (ssh is xterm)
if [ "$TERM" == "linux" ] ; then
  setfont /usr/share/consolefonts/Lat7-Terminus28x14.psf
  while :
  do
    ./padd.sh
    sleep 1
  done
  setfont /usr/share/consolefonts/Lat7-Terminus28x14.psf
fi
```

Obviously, you’d need to go through the fonts listed in /usr/share/consolefonts/ to determine which one is best suited for your needs (I just used the example font shown [here](https://unix.stackexchange.com/a/49823).)

closed 2018-07-08 by jpmck

lyttle123, 2018-11-05:
There is a better fix if your on a pi. Just pit these two lines in your config.txt file. 
framebuffer_width=480
framebuffer_height=320
Waveshare 5.0
[image]
I think it looks better to then changing the font. If anybody has the problem.