pi-hole/PADD #23: DNS over HTTPS

jonsy777, 2018-07-09:
**Is your feature request related to a problem? Please describe.**

When using cloudlflare's DNS over HTTPS, the built in "DNS Server" list in pi hole (web interface, config file, etc) goes blank. This leads to the list of DNS servers in PADD to appear blank.

**Describe the solution you'd like**
If there was a way to indicate that DNS over HTTPS was in fact being used, as well as indicate what DNS servers were being used within DNS over HTTPS, that would be awesome. 

**Describe alternatives you've considered**
I haven't really found a workaround/way to make this work. Because DNS over HTTPS is somewhat under the hood of pi-hole, I haven't really messed with it.

**Additional context**
I know it has a config file that contains the DNS on the end of DNS over HTTPS so that might be a way to grab the info there. (although i have minimal coding knowledge.)

p1r473, 2018-07-13:
I can't imagine the latency of DNS over HTTPS
Have you considered dnscrypt-proxy instead?

jpmck, 2019-02-10:
It is my understanding that using `cloudflared` will result in you using `127.0.0.1#5053` as your "upstream" DNS server. PADD checks for this and will display "Cloudflare" if this is the case.

closed 2019-02-10 by jpmck