pi-hole/PADD #24: Can you please have an option to print all blocked domains like the query log?

p1r473, 2018-07-13:
**Is your feature request related to a problem? Please describe.**
I;m always frustrated when a domain is blocked, and I check PADD, but I was too late, and its not the latest block anymore

**Describe the solution you'd like**
A query log like in the original chronometer, minus all of its bugs

**Describe alternatives you've considered**
tail -f /var/log/pihole.log | awk '/\/etc\/pihole\/gravity.list/ {if ($6 != "address" && $6 != "name" && $6 != "/etc/pihole/gravity.list") print $6; else;}'

**Additional context**
The problems the above statement had:
https://github.com/pi-hole/pi-hole/issues/717

p1r473, 2018-07-13:
Sidenote: great job on the project. Im using it and it works great.
Maybe have a "scrolling area" beneath the header that scrolls the latest?

p1r473, 2018-07-15:
See on the right screenshot where you have "Latest"?
It's basically a list of all the latest. Sort of if you do a tail of the log. So I guess same as on the left, but the domain only (no IPs)

For example:
pokemon.com
roku.com
microsoft.com

Just a list of all blocked domains... it helps make whitelisting easier

p1r473, 2018-07-20:
Let me try to rephrase the request so it's easier to understand
When I try to whitelist something, I need to see whats blocked. The query log on the pihole is terrible slow, so I usually do 
tail -f /var/log/pihole.log
to see what's being blocked
However, this has lots of junk in it. The original chronometer had a line in it:
tail -f /var/log/pihole.log | awk '//etc/pihole/gravity.list/ {if ($6 != "address" && $6 != "name" && $6 != "/etc/pihole/gravity.list") print $6; else;}'
This used to do vertical scrolling of all blocked domains. However, sometimes it would mess up and print the wrong word  (https://github.com/pi-hole/pi-hole/issues/717)

I'd really like, if under your big header, you could scroll a list of all blocked domains.
Right now you only show top and last domain. But, what if I want to see whats being blocked, but they are blocking so fast that I missed the last one, and now its a new last one?

jpmck, 2018-08-14:
>  Right now you only show top and last domain. But, what if I want to see whats being blocked, but they are blocking so fast that I missed the last one, and now its a new last one?

I would recommend using `pihole -tail`:

[image]

This color codes the output of of `/var/log/pihole.log`. I don't plan on adding a "ticker" of recently blocked domains to PADD.

closed 2018-08-14 by jpmck

p1r473, 2019-09-20:
The ticker would be a really cool feature, especially for when you want to find what needs whitelisting