pi-hole/PADD #25: PiHole Version Checking Still Broken

stephenjmcmahon, 2018-07-14:
PiHole Versioning check is still broken it seems after PADD v2.1.0

Removing PiHoleVersion then rebooting works but any subsequent reboot results in a component falsepositive or reporting the wrong version. 

Currently it's claiming PiHole 3.3.1 is out of date and on my other PiHole it's fine at the moment. Seems to be random as one of them claimed all versions (PiHole, Web, FTL) were out of date then after a reboot only PiHole was.

Unfortunately there are no errors or anything this time around like v2.0.0

stephenjmcmahon, 2018-08-07:
Had one Pi upgrade (v4.0 PiHole release) and reboot just fine showing all up to date. 2nd Pi showing v4.0 but has an update available for it. Removed PiHoleVersion file and restarted but no difference.

Is there a main bug thread for this I can post in? I see this is marked as duplicate

cheers

tgjohnst, 2018-08-07:
Mine did not report that my pihole was out of date despite the release, and after upgrade now reports the versions correctly as v4.0 but also displays the red out-of-date message.

tgjohnst, 2018-10-19:
I think I may have fixed this in my latest PR https://github.com/jpmck/PADD/pull/31

Basically, if the date of last check was equal to the current date at script startup, the script would never update today's date since it was after the check and thus would also never redo the version check. Version checking would only happen once the script was rerun again on a different day (since today's date is also updated at script startup).

jpmck, 2019-02-20:
The update checking function in PADD was completely written (see 17a158b2f85de28d7d11c98997839e7d4285d78e).

This should be completely resolved, I hope. Thanks @tgjohnst!

closed 2019-02-20 by jpmck