pi-hole/PADD #26: SetLocale Warnings

NobleAscent, 2018-07-19:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
PADD is unable to show System Stats and at the end there is twice a bash warning:
`bash warning: setlocaleL LC_ALL: cannot change locale (en_US.UTF-8)`

**To Reproduce**
Steps to reproduce the behavior:
1. wget -N https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh
2. sudo chmod +x padd.sh
3. nano ~/.bashrc
4. sudo reboot

**Expected behavior**
No errors at the end of screen.

**Screenshots**
[image]

**Additional context**
None

Andrew-H2O, 2018-08-17:
I encountered this too.

This fixed it;
Run _raspi-config_, then select **Localisation Options** -> **Change Locale**
Select to generate **en_US.UTF-8 UTF-8** , and step through the prompts.  It's not necessary to make this your default locale, just that it has been created.

jpmck, 2019-02-20:
The use of locales other than C was removed in 17a158b2f85de28d7d11c98997839e7d4285d78e.

This should be resolved.

closed 2019-02-20 by jpmck