pi-hole/PADD #32: PADD "Updates are Available" after updating - v2.1.0

vutorious, 2018-10-20:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
I've updated PADD using: wget -N https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh and I rebooted the Pi.  The screen still says Updates are Available and version 2.1.0.

**To Reproduce**
Steps to reproduce the behavior:
1. SSH into Raspberry Pi
2. Run: wget -N https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh
3. Reboot Pi
4. Profit...?

**Expected behavior**
After updating, I'd figure the message would read "System Healthy"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

tgjohnst, 2018-10-20:
I think I may have fixed this in PR https://github.com/jpmck/PADD/pull/31 , if you feel like testing it let me know if that fixes it

vutorious, 2018-10-23:
Unfortunately, I'm still getting the same error after making the update.  I've also tried removing the piHoleVersion, but after running pihole -up or rebooting the Raspberry Pi, it still shows "Updates are available".

Attached is my piHoleVersion file
[piHoleVersion.txt](https://github.com/jpmck/PADD/files/2504157/piHoleVersion.txt)

vutorious, 2018-10-23:
I've been able to reproduce the error I'm seeing.

1. Updated part of the script with the fix provided above
2. Run "pihole -up"
3. Reboot the Pi
4. The version numbers show up red and also show "Updates are available"
5. I SSH into the pi and run ./padd.sh - it updates the colors to green, but still shows "Updates are available"
3. I close the script with ctrl+c and run ./padd.sh once again, everything is correct (green and "System is healthy")

tgjohnst, 2018-10-25:
Thanks for the clear description of the error!! Do you have your PADD dashboard set to start up with the Pi? I do as well, and my version checking also fails during startup due to the pihole DNS not being fully up and running by the time the PADD script starts. A rerun usually fixes this, but I hadn't noticed the behavior in your step (5) which is curious. 

I'm thinking about how to better solve this - perhaps a flag set on script startup that bypasses the last-check date requirement and causes GetVersionInformation() function to be rerun until the version query to remote servers was actually successful (at the moment it looks like it's a simple "a"="b" check, but if the remote server doesn't answer sensically that is still counted as a non-match). @jpmck  any thoughts?

R3clu23, 2018-11-03:
I have this same issue, tried some of the other fixes from others

https://imgur.com/odpS2FK

shadshack, 2019-02-14:
I'm seeing this same problem on PADD v2.2.1. I added a "sleep 5" line to the ~/.bashrc file script before the "if [ "$TERM" == "linux" ] ; then" line, and now it's working.

jpmck, 2019-02-20:
I've completely re-written the version checking function in PADD (see 17a158b2f85de28d7d11c98997839e7d4285d78e).

This should be resolved.

closed 2019-02-20 by jpmck

shadshack, 2019-02-20:
You posted this reply about 10 minutes after I had just finished reinstalling the OS and setting back up my Pi-hole. I noticed it was working weird still and the version was 3.0. I must have grabbed PADD just before you pushed 3.0.1. Updated to that, and it's working 100% now! You rock!