pi-hole/PADD #36: Screen won't sleep....

Andymc1280, 2018-11-22:
Simply put, my display is permanently on, even after editing crontab and adding:

# PADD Sleep Turn off the PiTFT+ at midnight

00 00 * * * sh -c 'echo "0" > /sys/class/backlight/soc\:backlight/brightness'

# Turn on the PiTFT+ at 8:00 am
00 08 * * * sh -c 'echo "1" > /sys/class/backlight/soc\:backlight/brightness'
# PiTFT+ SLEEPY TIME Turn off the PiTFT+ at midnight
00 00 * * * sh -c 'echo "0" > /sys/class/backlight/soc\:backlight/brightness'
# Turn on the PiTFT+ at 8:00 am
00 08 * * * sh -c 'echo "1" > /sys/class/backlight/soc\:backlight/brightness'

Running latest Pihole (4.0) and PADD (2.1.0)

Andymc1280, 2018-11-22:
`PADD Sleep Turn off the PiTFT+ at midnight

00 00 * * * sh -c 'echo "0" > /sys/class/backlight/soc:backlight/brightness'
Turn on the PiTFT+ at 8:00 am

00 08 * * * sh -c 'echo "1" > /sys/class/backlight/soc:backlight/brightness'
PiTFT+ SLEEPY TIME Turn off the PiTFT+ at midnight

00 00 * * * sh -c 'echo "0" > /sys/class/backlight/soc:backlight/brightness'
Turn on the PiTFT+ at 8:00 am

00 08 * * * sh -c 'echo "1" > /sys/class/backlight/soc:backlight/brightness'`

jpmck, 2018-11-23:
This beyond the scope of this project.

(See Adafruit's [documentation](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/backlight-control) for more information on setting up your screen.)

closed 2018-11-23 by jpmck