pi-hole/PADD #39: PADD version file not found

ramlev, 2019-01-18:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
When trying to start PADD.sh, i'm  getting a lot of errors.

**To Reproduce**
Steps to reproduce the behavior:
1. start padd.sh

**Expected behavior**
Wanted to see correct info

**Screenshots**
[image]

**Additional context**
just installed my rpi with pihole.

tgjohnst, 2019-01-18:
Hi ramlev,

The version file won't be found until PADD has run successfully, but is not the root of the problem. It looks like this is a locale issue - the printf errors are because of your locale's NUMERIC settings, and other errors because you don't have en.us.UTF-8 installed as a locale for your shell. Here's the solution that should fix it:
https://github.com/jpmck/PADD/issues/26#issuecomment-413909951

jpmck, 2019-02-20:
The en.us.UTF-8 locale is no longer required for PADD (see 17a158b2f85de28d7d11c98997839e7d4285d78e).

This should be resolved.

(Thanks for looking into this @tgjohnst!)

closed 2019-02-20 by jpmck