pi-hole/PADD #41: bc is no longer installed by default in Pi-hole

chrisys, 2019-02-05:
Not a bug report as such by more of a PSA: in [this](https://github.com/pi-hole/pi-hole/commit/3035fbf468dacc41762ec91015f5f15bea5c33fb#diff-689793e68cde6270d074695b71c969af) commit, Pi-hole removed the installation of `bc` by default. To run PADD now you'll need to manually install it. I've added it to the [balena Pi-hole](https://github.com/klutchell/balena-pihole/pull/3) project.

jpmck, 2019-02-09:
Thanks Chris. I'm in the process of removing the use of `bc` for the next release.

jpmck, 2019-02-09:
Duplicate of [Various "command not found' error messages continually displayed while running PADD.sh](https://github.com/jpmck/PADD/issues/42).

closed 2019-02-09 by jpmck