pi-hole/PADD #42: Various "command not found' error messages continually displayed while running PADD.sh

peatrick, 2019-02-09:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*
```
- Gathering version information...
  - Pi-hole Core v4.2.1
  - Web Admin v4.2
  - FTL v4.2.1
  - PADD v2.1.0
- Checking for update information...
  - Pi-hole is up-to-date!
```

**Describe the bug**
```
START UP ===================================================
- Writing PID (22896) to file...
- Checking for PADD version file...
  - PADD version file found... deleting.
- Gathering system information...
./padd.sh: line 192: bc: command not found
./padd.sh: line 652: bc: command not found
./padd.sh: line 652: bc: command not found
./padd.sh: line 652: bc: command not found
./padd.sh: line 202: bc: command not found
./padd.sh: line 676: bc: command not found
./padd.sh: line 676: bc: command not found
- Gathering Pi-hole information...
./padd.sh: line 676: bc: command not found
- Gathering network information...
```

**To Reproduce**
Steps to reproduce the behavior:
1. Managed to bork my original Pi-hole / Unbound / OpenVPN server by upgraded to Buster/sid branch and doing a little too much tinkering, so i wiped the micro sd card last evening and reinstalled everything.  Seems to be working fine, but just attempted to add [PADD.sh](https://github.com/jpmck/PADD/) today and getting these odd errors when attempting to run it.  Have checked both the FAQ and Known issues sections, but didn't really find anything relevant to my issue.
> _Linux pihole2 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux_ running **Raspbian GNU/Linux 9.6 (stretch)**
2. Followed the simple instructions to install PADD.sh (i have it working on two other Pi-holes, but they are both Ubuntu Server VMs, this device is an actual **Raspberry Pi 3 B+** with Rapbian Lite image applied to sd card via [Rufus 3.4p](https://rufus.ie/)

**Expected behavior**
Expect PADD to start and display _Status: Active_

**Screenshots**
Working PADD instances, both Ubuntu 18.04.1 LTS VMs running **4.15.0-45** kernel:
[image]

NOT working PADD, on the RPi 3 B+ (again, with Pi-hole, Unbound & OpenVPN Server): 
[image]

**Additional context**
Not sure if I'm just missing some package, or if this is a really simple error I'm overlooking, but my expertise is obviously rather limited.  Last time I used PADD.sh on this same RPi i had _chronometer2.sh_ installed and used the alternate instructions to upgrade from that.  This was a fresh installation, **uptime =** ` 14:37:26 up 1 day,  5:52,  1 user,  load average: 0.03, 0.04, 0.01`

Thank you for the wonderful piece of software, I love running it on all three of my Pi-holes.  Typically just run two for primary / secondary nameservers (via Unbound) with **Conditional Forwarding** enabled, which points back to my Edgerouter 12.  Also have dnsmasq enabled on the edgerouter, not sure that is relevant, though.  Appreciate your time and attention to my little matter.  Thank you kindly, sir.

jpmck, 2019-02-10:
This should be resolved. Please [update PADD](https://github.com/jpmck/PADD/wiki/Updating-PADD).

closed 2019-02-10 by jpmck

peatrick, 2019-02-10:
Thank you kindly, sir.  Just removed / reinstalled -- **v2.2.0** now working wonderfully.
[image]