pi-hole/PADD #45: Upgrading from v2.1.0 to v2.2.1 Causes Resolution Issues with Script

stephenjmcmahon, 2019-02-19:
Upgraded from v2.1.0 to v2.2.1

Used to be displayed in Terminus 8x14 but after upgrading it cuts the top of the script and forces some stats to new lines, mid line. Followed upgrade instructions from main page.

Changing it to 6x12 does get it on screen but renders it almost unreadable at a glance - I also did not see any mention that this was intended for a future update.

https://i.imgur.com/1qIoJ8b.jpg

Steps to reproduce the behavior:
1. Update script

tgjohnst, 2019-02-19:
I am also experiencing this issue.

I just updated to the latest version of PADD.sh (2.2.1) and pihole (4.2.2). The second display line seems to be messed up in my default view - the lower half of the ASCII PADD logo is now cut off, and there is no line displaying the PADD version next to it. This is the case both on my pi TFT screen as well as via SSH. 

**Expected behavior**
Previous versions displayed the complete PADD logo regardless of version / version status, and displayed the PADD version to the right of it before the system status text

**Screenshots**
[image]

jpmck, 2019-02-20:
🤦‍♂️ 100% my fault. Thanks @stephenjmcmahon, @tgjohnst!

I got this cleaned up with v3.0.0!

closed 2019-02-20 by jpmck