pi-hole/PADD #47: PADD not starting

mikebetz42, 2019-02-19:
PADD not starting, and yes I have edited "~/.bashrc". it use to work, but it's no longer displaying on the Pi. I can run it in an SSH session without a problem.

Pi-hole® v4.2.2, Web v4.2, FTL v4.2.2, PADD v2.2.1

Linux 4.14.98-v7+ armv7l
Raspbian GNU/Linux 9.8 (stretch)

[image]

Thank you,

Mike

maserowik, 2019-02-19:
sudo chmod +x padd.sh  

 

 

I had to redo this command in order for it to work for me.

 

Mike

 

 <https://user-images.githubusercontent.com/25675511/53043566-99d47d80-343d-11e9-8459-0081ced83250.jpg> 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

mikebetz42, 2019-02-19:
I fixed my problem....  edited "~/.bashrc" and 

if [ "$TERM" == "linux" ] ; then

  **changed to**  

if [ "$TERM" == "xterm" ] ; then

Did the trick.

closed 2019-02-19 by mikebetz42

tspotz, 2019-02-20:
> I fixed my problem.... edited "~/.bashrc" and
> 
> if [ "$TERM" == "linux" ] ; then
> 
> **changed to**
> 
> if [ "$TERM" == "xterm" ] ; then
> 
> Did the trick.

since a SSH connection is also "xterm" this leads to PADD starting even on ssh'd terminal.
i changed this last night (after upgrading the pi) and encounter this problem to

if [ "$SSH_TTY" == "" ] ; then

locally loged in the SSH_TTY is empty.

stephenjmcmahon, 2019-02-24:
Sorry to re-open this but I've also run into the same issue using the latest PADD (v3.0.2) only after updating the kernel from 

```
pi@PiHole-02:~ $ uname -a
Linux PiHole-02 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

to

```
pi@pihole-01:~ $ uname -a
Linux pihole-01 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

I had to ```sudo raspi-config``` to re-enable auto-login to console and verified ~/.bashrc still had the included code to auto-run but still nothing.

as @tspotz mentioned above that was the trick to fixing this.

changing
`if [ "$TERM" == "linux" ] ; then`
to
`if [ "$SSH_TTY" == "" ] ; then`

R3clu23, 2019-03-04:
> since a SSH connection is also "xterm" this leads to PADD starting even on ssh'd terminal.
> i changed this last night (after upgrading the pi) and encounter this problem to
>
> if [ "$SSH_TTY" == "" ] ; [...]

I have the same problem, how did you edit the .bashrc file if you can't stop the script from running?

stephenjmcmahon, 2019-03-04:
Ctrl+C should stop PADD from running and drop you back into CLI

R3clu23, 2019-03-04:
> Ctrl+C should stop PADD from running and drop you back into CLI

It doesn't it just restarts the script over again.  

I edited my .bashrc with:

if [ "$TERM" == "xterm" ] ; then

tspotz, 2019-03-05:
press Ctrl-C several times since the .bashrc itself is a bash-script which should be exited (seen the while loop which restarts the padd.sh after waiting a second ?)
I connected directly with an USB Keyboard to RPi

R3clu23, 2019-03-05:
Okay,  pressing ctril-c while the script loaded did in fact cancel the script from loading and I'm back at the pi@pihole~$ CLI line.  Thanks for the help.

R3clu23, 2019-03-05:
> press Ctrl-C several times since the .bashrc itself is a bash-script which should be exited too (seen the while loop which restarts the padd.sh after waiting a second ?)
> I connected directly with an USB Keyboard to RPi

Okay, pressing ctrl-c while the script loaded did in fact cancel the script from loading and I'm back at the pi@pihole~$ CLI line. Thanks for the help.

timkingman, 2019-04-05:
When `raspi-config` turns on auto-login, it writes your current `$TERM` to `/etc/systemd/system/getty@tty1.service.d/autologin.conf`. Edit that file and change the last parameter to `linux`, and then the test in .bashrc should start padd.sh on only the console auto-login.