pi-hole/PADD #49: SyntaxError: invalid syntax

chriswike, 2019-02-21:
Trying to setup PADD on a brand new install of pihole, PADD Version 3.0.2

When trying to run padd.sh I get the following error:

`
  File "padd.sh", line 23
    today=$(date +%Y%m%d)
          ^
SyntaxError: invalid syntax
`
When I run  locale -va output

`
locale: en_GB.utf8      archive: /usr/lib/locale/locale-archive
-------------------------------------------------------------------------------
    title | English locale for Britain
   source | RAP
  address | Sankt Jørgens Alle 8, DK-1615 København V, Danmark
  contact | Keld Simonsen
    email | bug-glibc-locales@gnu.org
 language | British English
territory | United Kingdom
 revision | 1.0
     date | 2000-06-28
  codeset | UTF-8

locale: en_US.utf8      archive: /usr/lib/locale/locale-archive
-------------------------------------------------------------------------------
    title | English locale for the USA
   source | Free Software Foundation, Inc.
  address | http://www.gnu.org/software/libc/
    email | bug-glibc-locales@gnu.org
 language | American English
territory | United States
 revision | 1.0
     date | 2000-06-24
  codeset | UTF-8

locale: C.UTF-8         directory: /usr/lib/locale/C.UTF-8
-------------------------------------------------------------------------------
    title | C locale
    email | aurel32@debian.org
 language | C
 revision | 1.6
     date | 2016-08-08
  codeset | UTF-8
`

chriswike, 2019-02-21:
I was able to fix this by editing the `"~/.bashrc`

I changed 

`if [ "$TERM" == "linux" ] ; then`

to

`if [ "$TERM" == "xterm" ] ; then`

as referenced in this post [https://github.com/jpmck/PADD/issues/47#issuecomment-465298670](https://github.com/jpmck/PADD/issues/47#issuecomment-465298670)

Any ideas why the `SyntaxError: invalid syntax` ?

closed 2019-02-21 by chriswike

jpmck, 2019-02-22:
That is a mystery... I don't see any syntax error with that command at all... https://ss64.com/bash/date.html

timkingman, 2019-04-05:
> I was able to fix this by editing the `"~/.bashrc`
> 
> I changed
> 
> `if [ "$TERM" == "linux" ] ; then`
> 
> to
> 
> `if [ "$TERM" == "xterm" ] ; then`
> 
> as referenced in this post [#47 (comment)](https://github.com/jpmck/PADD/issues/47#issuecomment-465298670)

I think the root issue here is that auto-login is setting a term other than linux. Update the auto login script [like this](https://github.com/jpmck/PADD/issues/47#issuecomment-480329581)