pi-hole/PADD #51: Upgrading Kernel Breaks ~/.bashrc Code

stephenjmcmahon, 2019-02-24:
See: https://github.com/jpmck/PADD/issues/47

I was able to replicate this on both my PiHole installations by upgrading the kernel via `sudo apt-get update && sudo apt-get upgrade` . Both times they booted to login only then after `sudo raspi-config` to re-enable auto-login the just login normally without executing anything. PADD 3.0.2

Old kernel
```
pi@PiHole-02:~ $ uname -a
Linux PiHole-02 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

New kernel

```
pi@pihole-01:~ $ uname -a
Linux pihole-01 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

Only after changing [per @tspotz in the other thread]
`if [ "$TERM" == "linux" ] ; then`
to
`if [ "$SSH_TTY" == "" ] ; then`
in `~/.bashrc `was I able to get this to function properly again

alonchis, 2019-03-01:
I have a similar issue. when i echo $TERM directly in the pitft with a keyboard i get "xterm-256color" and not linux, so i edited that in and it works now (although i might need to put in a delay since it tries to check for for internet connection and fail for a while).

timkingman, 2019-04-05:
When `raspi-config` turns on auto-login, it writes your current `$TERM` to `/etc/systemd/system/getty@tty1.service.d/autologin.conf`. Edit that file and change the last parameter to `linux`, and then the test in .bashrc should start padd.sh on only the console auto-login.

derekschinke, 2019-09-16:
@timkingman neither of those worked for me.

dschaper, 2020-08-17:
Almost a year since last comment. Closed.

closed 2020-08-17 by dschaper