pi-hole/PADD #52: Route: command not found

hasecbinusr, 2019-03-07:
**Describe the bug**
Updated raspberry pi with apt-get update and apt-get upgrade and updated PADD to v. 3.0.2.  Now PADD generates the error `./padd.sh: line 263: route: command not found`.

**To Reproduce**
Steps to reproduce the behavior:
1. Update working pi-hole system.
2. Update PADD.

**Expected behavior**
Expected error-free performance as prior to upgrade.

**Screenshots**
[image]

**Additional context**
This may be fixed as discussed in #50.

hasecbinusr, 2019-03-08:
Changed line 263 as PR #50 commit and operation is error-free again.

jpmck, 2020-02-23:
Corrected in #50. Thanks @tom472. This was added to v3.1.

closed 2020-02-23 by jpmck