pi-hole/PADD #54: Pi hole is offline - but online

ramlev, 2019-04-02:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. ... padd.sh running on my RPI
2. ... from web interface to pi.hole, disable permanetely
3. ... check that pi hole is off-line in top of padd.sh
4. ... from web interface to pi.hole enable again
5... padd.sh keep showing 'pi hole is off-line'

**Expected behavior**
pi hole is online, but the padd.sh telling me it's offline, until i restart padd.sh

**Screenshots**
[image]

boarder2k7, 2019-06-10:
I am having this same issue on PiHole version 4.3

It will display as online upon first boot, but goes offline if PiHole is disabled, and never refreshes to back online. Ex: After using the "disable for 5 minutes" option

KnownAim, 2020-06-01:
Issue still exists. Can easily replicate on PADD v3.1. If PiHole is disabled and re-enabled, PADD will not update the information until it is manually restarted.

blurfl, 2020-12-19:
On my setup (v5.2.1), this seems to only affect the retro logo used by my 7" touchscreen. I run pihole from .bashrc, and padd.sh sees the screen size as 100x30, so chooses the 'mega' screen size.

 PrintLogo() uses a variable **full_status_** for the status message and checkmark in all except the retro logo. That uses **mega_status**, and mega_status doesn't seem to get updated.

 I've tried two approches to a patch, either seems to work:

by altering PrintLogo() to use fuil_status_  :
```
# use full_status_ with the retro_logo because mega_status doesn't reflect 'healthy' after a disable timeout
     CleanPrintf "${padd_logo_retro_3}   ${pihole_check_box} Core  ${ftl_check_box} FTL   ${full_status_}${reset_text}\e[0K\\n"
```

or by adding a line to NormalPADD() to update mega_status in the way used by several other screen sizes:
```
    pico_status=${pico_status_ok}
    mini_status_=${mini_status_ok}
    tiny_status_=${tiny_status_ok}   
    # update mega_status to reflect 'healthy' after disable timeout
    mega_status=${mega_status_ok}   
```

I don't know which approach would be best, and haven't tested either enough to know what other effects either change would have.

blurfl, 2020-12-20:
Checking the nice [custom PADD](https://gist.github.com/giuliomagnifico/beb9b3211b7a3580457b438bb4c9326d) from @guiliomagnifico mentioned [here](https://github.com/pi-hole/PADD/issues/119#issuecomment-703099520), the same issue occurs, and the same patches (either one) correct it.

KnownAim, 2022-02-19:
This can now be closed due to #167.

closed 2022-02-19 by yubiuser