pi-hole/PADD #56: PADD display shifted up wards and not returned to normal after long domain displayed

cmdshft, 2019-04-22:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
When a very long domain is displayed on PADD output, the entire CLI is shifted upwards. It does not return to a normal display after a new shorter domain is displayed

**To Reproduce**
Steps to reproduce the behavior:
1. have a long domain displayed

**Expected behavior**
Scroll long domain instead?

**Screenshots**
[image]

R3clu23, 2020-02-29:
I to am having the same issue with my display shifting up.  
I'm running:
Pi-Hole v4.4 [green]
Web v4.3.3 [green]
FTL v4.3.1 [green]
PADD v3.1 [red]

I have run the PiPadd update twice, not change.

pi@Pihole:~ $ wget -N https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh
--2020-02-29 13:36:04--  https://raw.githubusercontent.com/jpmck/PADD/master/padd.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 43103 (42K) [text/plain]
Saving to: ‘padd.sh’

100%[====================================>]  42.09K  --.-KB/s    in 0.04s

Last-modified header missing -- time-stamps turned off.
2020-02-29 13:36:04 (1009 KB/s) - ‘padd.sh’ saved [43103/43103]
              
[image]

yubiuser, 2022-08-20:
This is a quite old issue, but meanwhile it was solved. The display is still shifted with a very long domain but returns to normal if the next one is a "normal" sized domain.

[image]

closed 2022-08-20 by yubiuser