pi-hole/PADD #57: Custom PADD CLI resolution?

cmdshft, 2019-04-25:
Is there a way to define a custom layout for PADD on the Hyperpixel 4 display? I'd love to have the size of this:
[image]

But with the layout like this (logo, health status, versions at the top):

[image]

I have tried all the options in dpkg-reconfigure console-setup and nothing feels right.

cmdshft, 2019-05-07:
Any info on this?

cmdshft, 2019-05-20:
Bump. Is there no way to get the display to fit the "Mega" output to the 4" size?

jpmck, 2019-05-22:
I don't understand. You want a custom set-up based on the second photo of the setup that already exists?

cmdshft, 2019-05-22:
No, I want to use the Mega size in a way that will fill the entire display on the 4" Hyperpixel I have install.

[image]

jpmck, 2019-05-23:
Unfortunately I don't have any plans for developing any other sizes for PADD anytime soon, and developing a different one for each type of display that everyone is using would be close to impossible.

If you want, you can tweak the output as you see fit. The functions all follow the same naming convention: ```PrintSection``` (i.e. ```PrintNetworkInformation```). If you want to make a pull request to add your changes to the project, feel free.

closed 2019-05-23 by jpmck

myusuf3, 2020-02-12:
has anyone gotten this working? I am also running hyperpixel 4.