pi-hole/PADD #60: Autorun?

DJ-Anakin, 2019-06-02:
Using latest stable pi-hole. I would like PADD to start automatically. I'm nooby, and been messing with this for about an hour+ and can't get it. Right now outputting to old laptop screen, but I have a 3.5" TFT on order. 

Finally got it running using these steps, but, it asks for password in the middle of PADD startup. : https://old.reddit.com/r/pihole/comments/6pbjq5/auto_display_chronometer_on_lcd_screen_jessie/

like in this picture: https://image.ibb.co/f3e6LQ/Screen_Shot_2017_07_24_at_9_40_08_PM.png

I have this in VISUDO:
`pihole ALL=(ALL) NOPASSWD: /usr/local/bin/pihole -w *, /usr/local/bin/pihole -w -d *
`
Thanks.

cmdshft, 2019-06-02:
As per the wiki:

- Prevent the display from going to sleep by adding **consoleblank=0** to the end of **/boot/cmdline.txt**.
- Set your Pi to automatically log into the console using **raspi-config**.
- Configure the console to use the Terminus font at 8x14 by running **sudo dpkg-reconfigure console-setup**
- Set PADD to auto run by adding the following to the end of **~/.bashrc**:
`# Run PADD
# If we're on the PiTFT screen (ssh is xterm)
if [ "$TERM" == "linux" ] ; then
  while :
  do
    ./padd.sh
    sleep 1
  done
fi`

DJ-Anakin, 2019-06-02:
I already went through the wiki...

Sleep: Not worried about that right now...
raspi-config: already set
Terminus font: I don't see Terminus font... also, is this necessary?
~/.bashrc: Already done.

No change.

cmdshft, 2019-06-03:
Because you're using a different user than pi

You need to configure auto-login to console with user pi

DJ-Anakin, 2019-06-04:
Ah. That was it.. I wasn't using user Pi. Thanks.

DJ-Anakin, 2019-06-04:
I don't know why I closed this... It's still not working, but for different reason now. So I did a clean install of Raspbian and PiHole, I was having other problems. I've got a a 3.5" TFT now (this one: https://www.amazon.com/gp/product/B01IGBDT02/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1). It's displaying great. I've done everything in the install and any other steps I can find, and it will not start at boot. User is pi, i've got terminus font, i've got .bashrc set right, raspi-config is set to CLI and Pi user.

DJ-Anakin, 2019-06-04:
Ugh. No idea what I did different. I checked raspi-config AGAIN, and saved and rebooted and now it's working. There goes 2 hours i'll never get back. Oh well, at least it's working!

closed 2019-06-04 by DJ-Anakin

YamiYukiSenpai, 2021-01-16:
I'm using Ubuntu Server, and I haven't figured it out yet.