pi-hole/PADD #62: Error when started from r/o directory

stan-qaz, 2019-07-26:
*Before submitting a bug report, are you using a test version of Pi-hole? No, release version

**Describe the bug**
I get an error message under the usual text block that blinks in and out.

 CPU Temp: 104.4°F   CPU Load: 0.07, 0.02, 0.03    CPU Load: [··········] 1.8%  /home/pi/bin/padd.sh: line 536: ./piHoleVersion: Permission denied
/home/pi/bin/padd.sh: line 420: ./piHoleVersion: Permission denied

Steps to reproduce the behavior:
1. Change to /etc
2. Run PADD /home/pi/bin/padd.sh

**Expected behavior**
No error message

**Additional context**
If you cd to the pi home directory all is well.

tgjohnst, 2019-08-19:
Looks like this is due to the fact that the piHoleVersion file (used to check for updates) is created in the working directory, which in your case is read-only.

If @jpmck thinks it's appropriate, it would be pretty simple to change the script's behavior to always create the piHoleVersion file in the same directory as the piHole script (i.e. replace references to `piHoleVersion` and `./piHoleVersion` with `$(dirname "$0")/piHoleVersion` ), but I'm not sure this is a desired behavior so won't push a PR yet.

Bogdan107, 2021-12-16:
Proposal: create _piHoleVersion_ file under the `/tmp` firectory.

closed 2022-03-25 by yubiuser