pi-hole/PADD #63: Not auto Launching

Fiala06, 2019-08-14:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
It will not auto-launch on startup

I know it works as when I ssh into the machine it launches just fine. It just won't show up on my little 3.5 screen.

This is the screen I have: https://www.amazon.com/gp/product/B07N38B86S/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I've tried every combination of 
if [ "$TERM" == "linux" ] ; then
if [ "$TERM" == "xterm" ] ; then
if [ "$TERM" == "" ] ; then

All have the same outcome and rebooting each time. Any ideas? Spent to many hrs trying to get this to auto launch at this point. Looks great tho in putty :)

chriswike, 2019-08-14:
Did you set your Pi to automatically log into the console using raspi-config?

Fiala06, 2019-08-14:
Well thats one step closer! I chose the comamnd line and auto-login as pi. 

Now it just sits at the step below like it's still booting yet pihole is working and no PADD.

https://i.imgur.com/nkJ3sxA.jpg?1

chriswike, 2019-08-14:
Doesn’t look like you’ve selected the terminus font

’sudo dpkg-reconfigure console-setup’

8x14 is a good starting point 

There were kernel changes and how auto login to console is referenced in .bashrc. Do a update/upgrade

‘sudo apt-get update && sudo apt-get upgrade’

Do a sudo reboot after all this.

If that still doesn’t work, remove all mentions of PADD from your .bashrc file. See if the screen boots to a terminal with a cursor. If it doesn’t, you’ve got other problems.

Fiala06, 2019-08-14:
Thank you. It's working now. Not sure what exactly fixed it as I did both your suggestions 3 times and finally the 3rd time it took.

closed 2019-08-14 by Fiala06

elewin, 2020-12-04:
I ran into this exact same problem, and the fix for me was to comment out a line in /etc/rc.local that said `fbcp &`

This is the second time I've run into it -- the first time I found that fix and it worked, and just today after doing an apt-get update, it gave me the problem again except I forgot what I did to fix it. After several hours I finally stumbled across it again, and I'm pretty sure it will happen again at some point and I'll forget it yet again, so here's a note both for myself and anyone else who runs in to it in the future!