pi-hole/PADD #64: Set DNSSEC indicator color to non-red for local upstream resolver.

dschaper, 2019-09-20:
**Is your feature request related to a problem? Please describe.**
Currently DNSSEC is a red `Disabled` status when DNSSEC is not enabled. In some cases DNSSEC is not desirable and the red color gives the impression of an error state.

**Describe the solution you'd like**
If `/etc/pihole/setupVars.conf` shows 127.0.0.1 or similar for upstream then set the color to a neutral shade, it's neither good nor bad. Suggest blue as an indicator of nominal condition.

mwoolweaver, 2025-02-11:
[image]

can i extend this request to include `DHCP` as well

yubiuser, 2025-02-13:
@mwoolweaver 

Just a hint: Looking at your screenshot, it looks like you run PADD directly on your Pi-hole device. Did you know, with v6 you can even run it remotely from any other device which can connect to your Pi-hole.
Just run `./padd.sh --server IP_of_your_Pi-hole`

mwoolweaver, 2025-02-13:
How does the password work when 2FA is enabled?

yubiuser, 2025-02-13:
You need to create/use the 'app password' for all applications (including `PADD`) that don't implement 2FA.

mwoolweaver, 2025-02-14:
this should have closed with #411 or #412, right?

yubiuser, 2025-02-14:
Yes. It will autoclose when this is merged into master branch

yubiuser, 2025-02-28:
Mhh.. should have auto-closed, but obviously did not work

closed 2025-02-28 by yubiuser