pi-hole/PADD #66: Unsure of what is causing this to happen

cmdshft, 2019-09-28:
Every so often this has been appearing on the display of PADD. It goes away after the refresh of the screen but still not happy that this happens. How do I fix it?

[image]

sionide0278, 2019-10-27:
[image]

I get this issue too, it seems to be caused by the output of ps not being as expected. After a refresh, it goes away and the screen is as usual.

cmdshft, 2020-02-20:
BUMP

Any way to resolve this? Getting the feeling that the dev is disinterested in maintaining this lately.

tgjohnst, 2020-02-22:
@cmdshft what do you get if you run `pidof pihole-FTL` in terminal? The code checks to see if that's an empty string (and if so, avoids the ps call and says ftl is not running) but I'm wondering if that's returning a nonempty string that's still not a proper process id so the subsequent ps calls fail

AlexBarbulescu, 2020-07-20:
This is really annoying.

AlexBarbulescu, 2020-08-16:
I'm not that good at this but seems to work for me. 
Instead of verifying if there's any string we verify if the process exists.

search for `ftlPID=$(pidof pihole-FTL)`

replace the code if-else-fi

```
  if [ -n "$ftlPID" -a -e /proc/$ftlPID ]; then
    ftl_status="Running"
    ftl_heatmap=${green_text}
    ftl_check_box=${check_box_good}
    ftl_cpu="$(ps -p "${ftlPID}" -o %cpu | tail -n1 | tr -d '[:space:]')"
    ftl_mem_percentage="$(ps -p "${ftlPID}" -o %mem | tail -n1 | tr -d '[:space:]')"
  else
    ftl_status="Not running"
    ftl_heatmap=${yellow_text}
    ftl_check_box=${check_box_info}
    pico_status=${pico_status_ftl_down}
    mini_status_=${mini_status_ftl_down}   
    full_status_=${full_status_ftl_down}
    mega_status=${mega_status_ftl_down}
  fi
```

dschaper, 2020-08-16:
Might be related to some issues we found on the core repository code. 

https://github.com/pi-hole/pi-hole/pull/3599/files#diff-045722c02800a1422bcac00099aec6dbL27

Fix was to change `ps <PID>` to `pgrep -o "pihole-FTL"`

cosmith71, 2020-08-30:
I started over with a new flash of the latest Raspbian Buster (8/20/20), Pi-Hole, and PADD and the problem has not re-occured.

yubiuser, 2022-08-20:
Closing the issue because there was no activity > 2 years and I don't know how to reproduce the issue. Feel free to re-open or creating a new issue if necessary.

closed 2022-08-20 by yubiuser