pi-hole/PADD #67: add top client by blocked queries instead of top client by total queries

famesjranko, 2019-10-08:
I was wondering if you'd be able to add an option to see top client by blocked queries instead of top client by total queries?

Even if you simply added a line within padd.sh that one could uncomment to choose between the two different top client types.

I've looked through padd.sh, but not exactly sure where you're grabbing top client from to change this myself. It's obviously shown in the pi-hole web client, but haven't been able to track specifically where it's located to add myself.

tgjohnst, 2019-10-09:
It looks like GetFTLData() in PADD is calling the built-in `top-clients` endpoint from the pihole FTL api (see [docs](https://docs.pi-hole.net/ftldns/telnet-api/) here). Since this is built into the pihole API itself, it wouldn't be easy to change, and I don't see any available function that could be used to retrieve the data based on blocked rather than total queries (the only options available are how many clients to return and whether to include clients with zero requests). If it's something you really want, might be worth placing a feature request on the [pihole repo](https://github.com/pi-hole/FTL/issues) itself to add the ability to return top clients by blocked rather than total requests.

famesjranko, 2019-10-09:
Thanks for checking and explaining this. I might pass that onto the pihole repo and see where that goes! Cheers

jpmck, 2020-02-23:
As described by @tgjohnst, I'm not able to easily implement that using the Pihole FTL API.

closed 2020-02-23 by jpmck