pi-hole/PADD #68: Overfills screen..

Jeffmc1, 2019-10-17:
I have PADD Running on a 3.5" Pi TFT screen.. when it runs the TOP few lines scroll off (see pic) is there something I should change somewhere ?
[image]

Jeffmc1, 2019-10-17:
Although when running it in PUTTY (SSH On Win) it looks different....
[image]

ferazambuja, 2019-10-22:
Is the screen resolution. Try to make the putty screen smaller and you will see that it adapts.
You can find more info on the wiki https://github.com/jpmck/PADD/wiki/Sizes

jpmck, 2020-02-23:
The PADD Slim photo that you attached is exactly what the display should look like. No lines are being cut off. You can resize the font being used as mentioned by @ferazambuja and as described under the [Setup](https://github.com/jpmck/PADD/wiki/Setup) portion of the Wiki. (See the last two bullets.)

[image]

closed 2020-02-23 by jpmck

derekschinke, 2020-02-23:
FYI, I'm having an actual screen overfill issues with 3.1. Every font size I've tried results with the PADD logo being off screen, along with current version information.