pi-hole/PADD #69: Screen flashing (on update?)

landshrk, 2020-01-08:
PiHole v 4.3.2
FTL v 4.3.1
PADD v 3.0.2

**Describe the bug**
Just installed/changed to the official RPi Foundation's 7" touchscreen. When booting into console and straight into PADD, the display flickers. One flicker per instance, every few seconds, ranging from every ~2 to every ~10 seconds. The flickers are very short-lived, well under .25 seconds per.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the Official 7" touchscreen
2. Run PADD on it

**Expected behavior**
Have the screen not flicker, of course!

**Screenshots**
n/a

**Additional context**
This didn't happen previously, using the PiTFT 2.8" from Adafruit. It's only after swapping to this new screen that it's been going on.

I've tried playing with anything in /boot/config.txt and ~/.bashrc that seemed like it would have something to do with display parameters, but nothing has changed.

Also, this doesn't happen when booting into the DE, or when running just the console. Only when running PADD.

landshrk, 2020-01-08:
On closer inspection, it looks like it flashes roughly every 5.5 seconds, using the "one thousand one, one thousand two, ..." method of timing it.

tgjohnst, 2020-01-08:
I also have the same flicker (I use the pimoroni hyperpixel) and assumed this was the default refresh/redraw just doing its thing. Did your previous screen use the same layout (regular) or was it on the pico or mini layout?

landshrk, 2020-01-10:
> I also have the same flicker (I use the pimoroni hyperpixel) and assumed this was the default refresh/redraw just doing its thing. Did your previous screen use the same layout (regular) or was it on the pico or mini layout?

It would originally re-size down to the micro, but with some tweaking to the font and resolution, I would set it to the mini. And yeah, I assumed something similar re: the screen/display refresh. Just seems strange that it only manifests (so far in my experience) on this screen, as when in the normal DE, there's none of this present. Even if I were to open XTerm and run padd.sh, it doesn't do this. Only when the Pi is booted into the terminal-only environment and has PADD auto-start does it have this happen.

It's not the end of the world if it can't get fixed. I've already become accustomed to having it flicker next to me. It's mounted far enough outside my peripheral vision that it's not really an issue. But it would be nice to not have to do that, you know? And I know that if the display is sitting within my range of vision, it's very distracting.

tgjohnst, 2020-01-10:
Interesting that it's in the terminal-only mode but not the others - I wonder if there's a way to tweak the redraw rate in that mode? A cursory search doesn't find anything useful. 

I did notice that the main PADD loop directly prints each part of the display separately, sleeps 5, and then clears. So the clear is slow and then the screen sits cleared while the other functions are running and echoing, which is probably the perceptible black flicker. It'd be much quicker to have the SizeChecker and individual print functions run and print to a buffer string while the previous output is still visible, then flush both the clear and all the text to terminal at once\*. This would require minor changes to collect the output of the various print functions. I can take a crack at prototyping it next week and see if it improves this behavior.

\* e.g.
```
NormalPADD() {
 for (( ; ; )); do
  output=$(clear; GenerateOutput)
  echo "$output"`
 done
}
```
where GenerateOutput collects the outputs of the printing functions currently in the [NormalPADD](https://github.com/jpmck/PADD/blob/4cb0962b4f551c1bc06f2290c12a510f0da51a15/padd.sh#L1032) loop into a single string

@jpmck any thoughts?

Sekhan, 2020-01-22:
I have exactly the same problem on my 7" screen, PADD refresh every 5 seconds, causing the black flicker.

@tgjohnst Thanks for looking at it, let us know if you find a solution to fix it 👍

tgjohnst, 2020-01-23:
I'm not at home so can't test atm, but I pushed a first attempt where I simply moved the terminal clear to right before the print functions so there's less time it sits fully blacked out. If this doesn't notably improve things, I'll try the buffer string approach. If anyone wants to test it, https://github.com/tgjohnst/PADD/tree/faster_refresh_1

Sekhan, 2020-01-23:
@tgjohnst Thanks for your time ! I've tried it, but it return me this :
`./padd.sh : lign 7 : syntax error near unexpected token "newline"`
`./padd.sh : lign 7 : <!DOCTYPE html>`

tgjohnst, 2020-01-23:
Hi Sekhan, not sure why that is, did you try the raw file? https://raw.githubusercontent.com/tgjohnst/PADD/faster_refresh_1/padd.sh

Sekhan, 2020-01-24:
@tgjohnst My bad ! I didn't pay attention
I've run it, it seems that (on my screen) it's still flashing (at an irregular rate)

cmdshft, 2020-02-20:
This is not an issue with PADD, it's due to the command that PADD uses to clear the CLI display and prep for new info to be displayed. But surely there must be a better way to do this?

Sekhan, 2020-02-21:
https://unix.stackexchange.com/questions/81167/prevent-text-screen-blinking-when-doing-clear

I've found this interesting post, maybe it could help !

tgjohnst, 2020-02-21:
Thanks Sekhan, the buffering solution I suggested is in a comment in the second solution there (https://unix.stackexchange.com/questions/81167/prevent-text-screen-blinking-when-doing-clear#comment675237_126153) but I haven't yet had time to implement/test it. Maybe this weekend...

tgjohnst, 2020-02-22:
@cmdshft @Sekhan this should be fixed in my latest PR :) give it a try:
https://github.com/tgjohnst/PADD/blob/repaint/padd.sh

Sekhan, 2020-02-22:
@tgjohnst Perfect, it works for me ! Thank you :)

ludo453, 2022-10-23:
hello

on my side it's always flickering or flashing I see three lines displayed on the bottom.

did you make any changes to the script?

thank you

closed 2022-12-17 by github-actions[bot]