pi-hole/PADD #70: PADD screen turns off after 24 hours.

TheRealJperk, 2020-01-10:
**Describe the bug**
Running latest version of both PiHole and PADD on Raspian. I feel like I’ve tried everything I could find on google. 

After around 24 hours on, the screen turns off and I have to reboot to turn it back on. Once on, works perfectly. 

Using a Adafruit 3.5 piTFT screen on a Pi3+. 
Any ideas?

**To Reproduce**
Steps to reproduce the behavior:
1. Leave PADD running for ~24 hours.

**Expected behavior**
Screen to stay on (sleep from midnight to 8am), but goes black. Must reboot to see PADD.

tgjohnst, 2020-01-10:
Have you tried preventing the display from going to sleep by addding ‘consoleblank=0’ to the end of /boot/cmdline.txt

TheRealJperk, 2020-01-11:
No, I don’t believe so. Coded it in tonight. Will see if it happens again
and report back.

TheRealJperk, 2020-01-11:
Nope still turns off.

dschaper, 2020-08-17:
You'll need to ask the AdaFruit support for directions on how to keep the display working. That's your terminal and OS's responsibility.

closed 2020-08-17 by dschaper