pi-hole/PADD #71: Help with PADD

savage302, 2020-01-12:
Can some one help me on this, I have the for Raspberry Pi 4 Touch Screen with Case, 3.5 inch Touchscreen with Fan, 320x480 Monitor TFT LCD Game Display from amazon. I installed the drivers and ran the commands for PADD from githud, set to boot to CLI  and log in, but I just get what looks to be the last bit of booting lines stuck on the screen. (where it has "OK" in green on the left)

[image]

. Any help would be awesome.

knightphoenix, 2020-02-21:
hey,

just to confirm is your default terminal called Linux?
have you set padd to auto run on start?

moegrease, 2020-03-22:
hey I have run into the same issue (with a 4 inch screen 800x480) i have my pihole up and running and showing no issues. I have set to boot to CLI and log in automatically. I have also set it auto run on start with the code:

 if [ "$TERM" == "linux" ] ; then
  while :
  do
    ./padd.sh
    sleep 1
  done
fi

any help would be greatly appreciated

savage302, 2020-03-22:
I fixed my issue but looking up the drivers for the screen I was using.

moegrease, 2020-03-22:
Thanks for the quick response, thats the first thing i did last week before setting up pihole. [http://www.lcdwiki.com/4inch_HDMI_Display-C](url)

dschaper, 2020-08-17:
Issue reported as fixed.

closed 2020-08-17 by dschaper

elewin, 2020-12-04:
Try commenting out a line in /etc/rc.local that says `fbcp &`