pi-hole/PADD #72: instalation

SirGamer01, 2020-02-03:
i have folloed the guide privided but have failed to get my dispay to output anything other than a just a white screen and the cl have been just spiting out an error each time i put the code in to enable the display
though im not sure if its due to useing 2.8" PiTFT screen or not 
im quite new to linx and am unsure what to do to resolve this issue

knightphoenix, 2020-02-21:
Hi. 
What's the error you are seeing?
Are you able to get any output from the pi to the display?
what brand /model of display are you using?

SirGamer01, 2020-03-01:
Apolages for the slow reply,
OS: raspbian buster
Hardware: Pi 3b

What's the error you are seeing?
can quite remeber as iv not look in soem time. once i have time ill try to relook at it and find out what is was.

Are you able to get any output from the pi to the display?
the only out put that i am getting from the screen is a blank while screen
 
what brand /model of display are you using?

A: im using Adafruit 2.8" PiTFT Plus display that came with the kit below. 
https://thepihut.com/products/adafruit-ad-blocking-kit-for-pi-hole-with-2-8-pitft-no-soldering-ada3974

SirGamer01, 2020-04-04:
> 
> 
> Hi.
> What's the error you are seeing?
> Are you able to get any output from the pi to the display?
> what brand /model of display are you using?

so iv just had a chance to get back in to it and it was complete user error ...
appolagess for rasing an issue the was of my own doing.
btw now its working its fantastic :D

closed 2020-04-04 by SirGamer01