pi-hole/PADD #73: PADD reporting new version, despite already being the current version

cmdshft, 2020-02-20:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
PADD is reporting an update available, labeled v3.0.2 as red and says [!] Updates are available, despite 3.0.2 being the current (at the time of this report) version.

**To Reproduce**
Run PADD

**Expected behavior**
Obvious.

**Screenshots**
[image]

knightphoenix, 2020-02-21:
This is most likely being caused by the version number not being updated on the last commit. This can be checked by trying to update. Does updating fix the issue?

cmdshft, 2020-02-21:
I have done that and it's not resolved.

danozdotnet, 2020-02-28:
#493      padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')

The grep is looking for 'Location' when infact it's now 'location', the case changed. Secondly, no need to run grep then awk, just do it in one:

curl -sI https://github.com/jpmck/PADD/releases/latest | awk -F '/' '/location/ {print $NF}' | tr -d '\r\n[:alpha:]'

sionide0278, 2020-03-02:
> #493 padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')
>
> The grep is looking for 'Location' when [...]

Replicated on v3.1 and fixed by changing Location to location

PR #111 (2020-07-12): Use GitHub API for remote version detection.
Use the GitHub API for remote version detection. 

Bumps to version v3.2.2.

Closes #73, closes #82, closes #90, closes #107 
Resolves #110, resolves #97, resolves #73, resolves #80, resolves #87
closed 2020-07-12 by dschaper