pi-hole/PADD #76: Latest PADD added line breaks on PiTFT 3.5" - some content beyond range of screen

aliastechnology, 2020-02-24:
Updated Pi-hole and PADD this morning.

New PADD now has extra line breaks below each heading (FTL ===, NETWORK ===, DNS === etc.) causing the range of output to go beyond the screen resolution of the 3.5" PiTFT.
Logo, version and health information is no longer displayed as a result.

To Reproduce:
Update Pi-hole and PADD. Reboot.

Expected Behaviour:
PADD logo, version info and health status should be displayed at top of PiTFT 3.5"

Screenshot: 
[image]

Fede068, 2020-02-24:
Have exact the same problem after the Update!

dw808303, 2020-02-24:
I’ve reproduced this on my installation as well.

famesjranko, 2020-02-24:
confirmed for me as well.  I found editing the script and removing an "=" from each of the lines before new added blank lines fixed the issue, but can't understand why.

derekschinke, 2020-02-24:
Exact same issue here, for every font size that I have tried.

jpmck, 2020-02-24:
Noticed the same thing when I checked on my PADD, which resides in the basement.

I'm working on a fix for this that should be deployed in the next couple of days.

tgjohnst, 2020-02-24:
Sorry for introducing a bug! Didn't happen when I tested the various sizes but it looks like this is happening when the screen size is exactly the length of the header lines. I'm thinking the additional escape character to clear to end of terminal may count as another drawn character and push the line over the limit so it wraps to a newline?

Was thinking as part of my previous PR about dynamically drawing the == signs to screen length rather than using fixed #s, but didn't get that far...

chrisjeb, 2020-02-24:
Hi,

Tnx for all your hard work but i seems to have same issue

Kentsson, 2020-02-28:
Same issue here

ChappyEight, 2020-02-29:
Seems to be relatively universal.  Same issue here after updating tonight.

todrees1967, 2020-02-29:
Same issue here as well.

R3clu23, 2020-02-29:
Same issue.
[image]

sgu222e, 2020-02-29:
Adding my name to the list of people affected.

vekzla, 2020-03-01:
Same here also reporting padd is out of date even though I have updated to latest.

Kentsson, 2020-03-01:
Try removing the piHoleversion file (not sure about the name since I can't check it out now)

vekzla, 2020-03-01:
> Try removing the piHoleversion file (not sure about the name since I can't check it out now)
> […](#)
> -- Sent from my Android device with K-9 Mail. Please excuse my brevity.

Thank you for your reply, I have removed the piholeVersion file (you got the name correct) and rebooted the raspi, still showing updates available. See pic below of my issues (screen stretched similar to others above and still showing updates are available) 

[image]

thank you for your assistance

sgu222e, 2020-03-03:
So, I found a hackish fix for this.
Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to the next line to display blank.
lines changed:  636, 646, 653, 673, 704, 715, 722, 748 and 759

Hope that helps someone.

vekzla, 2020-03-05:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

Can Confirm this has worked on 7inch TFT touch screen, also fixed the update issue by following https://github.com/jpmck/PADD/issues/87

krispayne, 2020-03-05:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

This also resolved the issue on my 3.5" display

Fede068, 2020-03-06:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

This fix works perfect also for my display adafruit PiTFT+ 3.5". Thank you!

ChappyEight, 2020-03-16:
Apologies as I'm a bit of a RPi novice.  Aside from the hack posted above, has there been an official fix posted?

accordxtc, 2020-03-18:
I would like to know as well for a notice how we can go about doing the above hack. Thanks in advance

VeryColdAir, 2020-03-25:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

I change those lines and then rebooted. I saw that the line that started with DNS still had an empty line below it.

I changed delete a "=" from line 649 and it fixed it.

VeryColdAir, 2020-03-25:
> I would like to know as well for a notice how we can go about doing the above hack. Thanks in advance

If you don't have ssh enabled on your raspberry pi, take a look at [the official documentation for how to enable it](https://www.raspberrypi.org/documentation/remote-access/ssh/). (Skip to step 3 on the list there.)  If you don't want to use ssh, then skip to the second method on the post.

If you have one of the latest versions of windows 10, you can run `ssh pi@<ip address>` to ssh into the pi, no third-party software like putty required anymore. (Might have to enable first party ssh tool, but for me it was there by default).

==========================

One of the easiest ways I know of, _that avoids the use of a terminal text-editor_, is to get the NppFTP addin for NotePad++. This assumes you had ssh set up.

Within the NppFTP window, you can set up a profile for the raspberry pi running pi hole. 

1. Use the IP address for the host name
2. Set the connection type to SFTP (port will default to 22, which is what it'll need to be unless you changed the port)
3. Enter the device's username and password (default u: pi p: raspberry)
4. Set the initial remote directory to: "/home/pi/".

When you first connect, it will show a hash key(?) and ask if you want to accept/trust it. Hit yes and then on the file explorer panel below the profile and connection buttons, you should see the subdirectories of the /home/pi/ directory as well as the files within /home/pi/, include padd.sh. Double click on padd.sh to open it. Once it's open, change the lines that @sgu222e mentioned, and you should be good to go.

==========================

If you'd rather edit using the terminal only, then either use ssh to access the pi's terminal or plug a keyboard & mouse in to do it directly. I believe if you're going to plug a keyboard and mouse in, you can press ctrl+c to stop padd and give you control of the terminal. (Not 100% sure that will work.)

When I did it, I was already in the /home/pi/ directory, but if you want to make sure you're there, you can run the command:
`cd /home/pi/`

To see everything in the directory, run the command:
`ls`

You should see padd.sh listed. To edit it, run the command:
`nano padd.sh`

You'll then be able to go though the file and make changes to it. For the commands at the bottom, press ctrl and the character show to activate the command. (^x means ctrl+x)

astronyu, 2020-04-03:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

Thanks for sharing. I have no idea how to use CleanEcho command so went into editor and change it manually. It works!

dawnchorus, 2020-05-07:
> So, I found a hackish fix for this.
> Remove 1 = from the CleanEcho command that displays the name of each section for the regular/slim display. It seems like the 80th character is causing a wrap to [...]

Sorry, this must be fairly basic but how can I navigate to those exact lines? I can get through to the directory but there is no way I can see what number line I am looking at.

astronyu, 2020-05-07:
> Sorry, this must be fairly basic but how can I navigate to those exact lines? I can get through to the directory but there is no way I can see what number line I am looking at.

If you're using vi editor, follow this guide: https://kb.iu.edu/d/afcx
For nano, ctrl+_ will ask you the line to go to.

dawnchorus, 2020-05-07:
> If you're using vi editor, follow this guide: https://kb.iu.edu/d/afcx
> For nano, ctrl+_ will ask you the line to go to.

Thanks! It's better now but still not working all the way. I was able to navigate to the lines to delete the extra "="s from lines 636, 646, 653, 673, 704, 715, 722, 748 and 759 as well as 649 pointed out by "VeryColdAir"

The first line "PI-HOLE=====..." now shows up, however the PADD logo and version #, system healthy, etc. info still is not showing. Any ideas?

[image]

astronyu, 2020-05-07:
I've uploaded my padd.sh to pastebin **[here](https://pastebin.com/zZjHy45F)**

dawnchorus, 2020-05-09:
> I've uploaded my padd.sh to pastebin **[here](https://pastebin.com/zZjHy45F)**

thanks for this! Works perfectly now

FlappingBerd, 2020-06-26:
> Can Confirm this has worked on 7inch TFT touch screen, also fixed the update issue by following #87

I cant seem to get my 7" screen to fill completely - only around 80% is used. Any idea on how I may be able to maximize the use? For reference I am running a the Raspberry Pi 7" touchscreen.

derekschinke, 2020-07-03:
Updated,  and still have the exact same issue.

dschaper, 2020-07-12:
Can I get some values from `tput lines` and `tput cols` to see what the display values are?

sgu222e, 2020-07-15:
> Can I get some values from `tput lines` and `tput cols` to see what the display values are?

3.5" 
pi@pihole:~ $ tput lines
24
pi@pihole:~ $ tput cols
80

mycomachine, 2020-09-09:
whats that lovely blue color

yubiuser, 2022-08-20:
Closing this issue because there was no activity >2 years and a lot of new released of `PADD` were made.

closed 2022-08-20 by yubiuser