pi-hole/PADD #77: abnormally high cpu usage with latest update 3.1

famesjranko, 2020-02-24:
**Describe the bug**
Noted abnormally high cpu usage with latest update - +90%

**To Reproduce**
Steps to reproduce the behavior:
1. ... pull latest build
2. ... run padd.sh

**Expected behavior**
padd to execute and run as previous builds. 

**Additional context**
reverted back to previous build and cpu usage returned to normal.
running non-beta pihole (v4.3.3) on a raspberry pi zero.

robogeek78, 2020-03-30:
I am seeing something similar, However, /proc/loadavg is reporting upwards of .9 to 1.5.

If I use iostat, i am seeing more like 30% loading.

dschaper, 2020-08-17:
Still seeing this issue?

robogeek78, 2020-08-19:
I just checked and I am no longer seeing this issue.

closed 2020-08-19 by dschaper