pi-hole/PADD #79: PADDmini Load/Memory alignment issue

christianboyle, 2020-02-25:
Not sure if [this commit](https://github.com/jpmck/PADD/commit/94483123481abcd5c61fd9dc2dd8902506e933b7#diff-f0d2b00f20028b2c7f9febd84b9659afL756) has anything to do with it, but it seems like the [latest update (3.1)](https://github.com/jpmck/PADD/releases/tag/v3.1) has unset the indentation of the `Load` and `Memory` labels in PADDmini as well as the alignment of the `Load` and `Memory` bars.

Post-update:
[image]

Pre-update:
[image]

Profesyor, 2020-02-25:
And me wanting PADDmini to use my WHOLE 2.8 screen.

christianboyle, 2020-03-02:
Fixed here: https://github.com/jpmck/PADD/pull/83

christianboyle, 2020-09-01:
This will be fixed in an [upcoming release](https://github.com/pi-hole/PADD/pull/118).

closed 2020-09-01 by christianboyle