pi-hole/PADD #80: Support PADD failing to retrieve version

Kerdic, 2020-02-28:
The code currently has logic to handle failure to retrieve current versions of pihole, web, and FTL, and changes the text to yellow in those cases. However, the code assumes there will never be failure retrieving the PADD version.

Support failing to retrieve PADD version, and change text to yellow, instead of indicating that PADD is out of date.

Bellow are the changes I made to the code, all marked with # custom start and # custom end comments.

```
GetVersionInformation() {
  # Check if version status has been saved
  if [ -e "piHoleVersion" ]; then # the file exists...
    # the file exits, use it
    # shellcheck disable=SC1091

    # Today is...
    today=$(date +%Y%m%d)

    # was the last check today?
    if [ "${today}" != "${last_check}" ]; then # no, it wasn't today
      # Remove the Pi-hole version file...
      rm -f piHoleVersion
    fi

  else # the file doesn't exist, create it...
    # Gather core version information...
    read -r -a core_versions <<< "$(pihole -v -p)"
    core_version=$(echo "${core_versions[3]}" | tr -d '\r\n[:alpha:]')
    core_version_latest=${core_versions[5]//)}

    if [[ "${core_version_latest}" == "ERROR" ]]; then
      # Custom Start
      # core_version_latest=${core_version}
      # Custom End
      core_version_heatmap=${yellow_text}
    else
      core_version_latest=$(echo "${core_version_latest}" | tr -d '\r\n[:alpha:]')
      # is core up-to-date?
      if [[ "${core_version}" != "${core_version_latest}" ]]; then
        out_of_date_flag="true"
        core_version_heatmap=${red_text}
      else
        core_version_heatmap=${green_text}
      fi
    fi

    # Gather web version information...
    read -r -a web_versions <<< "$(pihole -v -a)"
    web_version=$(echo "${web_versions[3]}" | tr -d '\r\n[:alpha:]')
    web_version_latest=${web_versions[5]//)}
    if [[ "${web_version_latest}" == "ERROR" ]]; then
      # Custom Start
      # web_version_latest=${web_version}
      # Custom End
      web_version_heatmap=${yellow_text}
    else
      web_version_latest=$(echo "${web_version_latest}" | tr -d '\r\n[:alpha:]')
      # is web up-to-date?
      if [[ "${web_version}" != "${web_version_latest}" ]]; then
        out_of_date_flag="true"
        web_version_heatmap=${red_text}
      else
        web_version_heatmap=${green_text}
      fi
    fi

    # Gather FTL version information...
    read -r -a ftl_versions <<< "$(pihole -v -f)"
    ftl_version=$(echo "${ftl_versions[3]}" | tr -d '\r\n[:alpha:]')
    ftl_version_latest=${ftl_versions[5]//)}
    if [[ "${ftl_version_latest}" == "ERROR" ]]; then
      # Custom Start
      # ftl_version_latest=${ftl_version}
      # Custom End
      ftl_version_heatmap=${yellow_text}
    else
      ftl_version_latest=$(echo "${ftl_version_latest}" | tr -d '\r\n[:alpha:]')
      # is ftl up-to-date?
      if [[ "${ftl_version}" != "${ftl_version_latest}" ]]; then
        out_of_date_flag="true"
        ftl_version_heatmap=${red_text}
      else
        ftl_version_heatmap=${green_text}
      fi
    fi

    # PADD version information...
    padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | grep 'Location' | awk -F '/' '{print $NF}' | tr -d '\r\n[:alpha:]')

    # is PADD up-to-date?
    # Custom Start
    if [[ "${padd_version_latest}" == "" ]]; then
      padd_version_heatmap=${yellow_text}
    else
    # Custom End
      if [[ "${padd_version}" != "${padd_version_latest}" ]]; then
        padd_out_of_date_flag="true"
        padd_version_heatmap=${red_text}
      else
        padd_version_heatmap=${green_text}
      fi
    # Custom Start
    fi
    # Custom End

    # was any portion of Pi-hole out-of-date?
    # yes, pi-hole is out of date
    if [[ "${out_of_date_flag}" == "true" ]]; then
      version_status="Pi-hole is out-of-date!"
      version_heatmap=${red_text}
      version_check_box=${check_box_bad}
      pico_status=${pico_status_update}
      mini_status_=${mini_status_update}
      full_status_=${full_status_update}
      mega_status=${mega_status_update}
    else
      # but is PADD out-of-date?
      if [[ "${padd_out_of_date_flag}" == "true" ]]; then
        version_status="PADD is out-of-date!"
        version_heatmap=${red_text}
        version_check_box=${check_box_bad}
        pico_status=${pico_status_update}
        mini_status_=${mini_status_update}
        full_status_=${full_status_update}
        mega_status=${mega_status_update}
      # else, everything is good!
      else
        version_status="Pi-hole is up-to-date!"
        version_heatmap=${green_text}
        version_check_box=${check_box_good}
        pico_status=${pico_status_ok}
        mini_status_=${mini_status_ok}
        full_status_=${full_status_ok}
        mega_status=${mega_status_ok}
      fi
    fi

    # write it all to the file
    echo "last_check=${today}" > ./piHoleVersion
    {
      echo "core_version=$core_version"
      # Custom Start
      echo "core_version_latest=$core_version_latest"
      # Custom End
      echo "core_version_heatmap=$core_version_heatmap"

      echo "web_version=$web_version"
      # Custom Start
      echo "web_version_latest=$web_version_latest"
      # Custom End
      echo "web_version_heatmap=$web_version_heatmap"

      echo "ftl_version=$ftl_version"
      # Custom Start
      echo "ftl_version_latest=$ftl_version_latest"
      # Custom End
      echo "ftl_version_heatmap=$ftl_version_heatmap"

      echo "padd_version=$padd_version"
      # Custom Start
      echo "padd_version_latest=$padd_version_latest"
      # Custom End
      echo "padd_version_heatmap=$padd_version_heatmap"

      echo "version_status=\"$version_status\""
      echo "version_heatmap=$version_heatmap"
      echo "version_check_box=\"$version_check_box\""

      echo "pico_status=\"$pico_status\""
      echo "mini_status_=\"$mini_status_\""
      echo "full_status_=\"$full_status_\""
    } >> ./piHoleVersion

    # there's a file now
  fi
}
```

danozdotnet, 2020-02-29:
https://github.com/jpmck/PADD/issues/73#issuecomment-592382729

^ it's failing to retrieve because of case sensitivity.

draik, 2020-03-02:
The `grep 'Location'` on line 493 is the problem, as it should be "location" (lowercase L), but this can also be grouped with the awk that follows, as such:

`awk -F '/' '/location/ {print $NF}'`

Kerdic, 2020-03-02:
Thanks, I still think it is proper to condition for failure when retrieving the version, and show yellow text. I also find it useful to output the latest version to the outfile generated during version checks.

PR #111 (2020-07-12): Use GitHub API for remote version detection.
Use the GitHub API for remote version detection. 

Bumps to version v3.2.2.

Closes #73, closes #82, closes #90, closes #107 
Resolves #110, resolves #97, resolves #73, resolves #80, resolves #87
closed 2020-07-12 by dschaper