pi-hole/PADD #86: Fit PADDmini propely into adafruit 320x240 2.8 TFT screen.

Profesyor, 2020-03-04:
Hi, please fit PADDmini into  320x240 2.8 or resize the full version for the 2.8" screen.
[image]

Kerdic, 2020-03-11:
Not sure if this helps any, but I use the adafruit 320x240 screen, and ended up customizing the logic to better utilize the screen. (Note: my bezel blocks some of the first character, so this script shifts the section headers to the right one space also). And after this 3.1 update to stop flickering, I have noticed some artifacts (random characters at the end of the lines) I need to identify and clean up.

Anyway, here it is if it helps anyone. All modifications are marked in blocks of custom start and custom end. It would be nice if there was official support to better utilize that screen size however.

[image]

[padd.txt](https://github.com/jpmck/PADD/files/4320681/padd.txt)

Profesyor, 2020-03-11:
I have the same bezel, look at all that info THANK YOU!!!!!!!

Kerdic, 2020-03-11:
No problem. I save a customized version out of the way, since updating PADD replaces the padd.sh file entirely. Which means, after every PADD update, I manually work in my custom changes to the new version, and attempt to update the custom with new enhancements.

Profesyor, 2020-03-11:
Pfff. I'm getting ./padd.sh Permission denied

Profesyor, 2020-03-11:
Fixed

Profesyor, 2020-03-13:
Keep me posted. The random characters at the end of the lines, I got a 1> today. Couldn't this be due to the X amount of = in the NETWORK and the SYSTEM line?

Kerdic, 2020-03-13:
Updated
[padd.txt](https://github.com/jpmck/PADD/files/4330027/padd.txt)

I implemented the relevant changes from these two pull requests into the custom section, I think that should fix the random characters and improve the version checking. Let me know if you still get them:
https://github.com/jpmck/PADD/pull/90
https://github.com/jpmck/PADD/pull/85

Profesyor, 2020-03-13:
Fantastic works like a dream!

carltonwb, 2020-04-27:
Pardon my newness but I saved the old padd.sh file to padd.sh.old, I then uploaded the padd.txt and changed the name to padd.sh.
Now what do I do. I did a sudo pihole -up and it said all was fine.
Still shows it saying update available.
Thanks

Kerdic, 2020-04-27:
Yeah, the pihole -up downloads the latest padd.sh, so this customized/unofficial padd.sh that you put in place was replaced with the newest official padd.sh.

So, you should be able to repeat the steps, and instead of doing pihole -up, just restart to pick up the new script.

carltonwb, 2020-04-27:
Thank you.
Did a reboot but did not work.
Did a shutdown and restart and that worked fine.
Appreciate it.

kokein, 2020-05-01:
> Yeah, the pihole -up downloads the latest padd.sh, so this customized/unofficial padd.sh that you put in place was replaced with the newest official padd.sh.
> 
> So, you should be able to repeat the steps, and instead of doing pihole -up, just restart to pick up the new script.

Hi, this is great ! Can I buy You a coffe :) ? Do you have any support link ?

mctehlers, 2020-05-11:
> Pardon my newness but I saved the old padd.sh file to padd.sh.old, I then uploaded the padd.txt and changed the name to padd.sh.
> Now what do I do. I did a sudo pihole -up and it said all was fine.
> Still shows it saying update available.
> Thanks

Did the same thing and a shutdown and reboot. But I'm just getting permission denied errors.. What should I do? Chmod xx?

Profesyor, 2020-07-06:
> Updated
> [padd.txt](https://github.com/jpmck/PADD/files/4330027/padd.txt)
> 
> I implemented the relevant changes from these two pull requests into the custom section, I think that should fix the random characters and improve the version checking. Let me know if you still get them:
> #90
> #85

New Update

Profesyor, 2020-07-06:
@Kerdic 
New Padd Update

Profesyor, 2020-07-06:
https://github.com/jpmck/PADD/releases/tag/v3.2.1
Latest version

Kerdic, 2020-07-07:
Here is the custom I have with 3.2.1.

One user mentioned they would prefer the memory bar to line up with the rest, so I made that change as well. However, the display is shifted one line at initial boot until the first update. Still looking into that... but here is what I have at the moment.

[padd.txt](https://github.com/jpmck/PADD/files/4887038/padd.txt)

Profesyor, 2020-07-08:
@Kerdic Thanks, it works as always! I see the shifted line at the bottom but, meh you'll figure it out. He said this was his last update, I guess it's up to "us" now to keep it working like a charm, as always.

KnownAim, 2020-07-14:
> @Kerdic Thanks, it works as always! I see the shifted line at the bottom but, meh you'll figure it out. He said this was his last update, I guess it's up to "us" now to keep it working like a charm, as always.

Well, the repository was moved into pi-hole's so it looks like the devs there are continuing maintenance. Looks like as of now, a lot of the bugs have been squashed, but there is always room for improvement, such as this.

dschaper, 2020-07-14:
Heyup, we have moved PADD over to the Pi-hole namespace with the intention of keeping things going.

Profesyor, 2020-07-16:
Aaaand it's broken now!

dschaper, 2020-07-16:
PADD is fine, it's doing exactly what it should.

Profesyor, 2020-07-16:
I've seen it. Pi-hole dashboard is broken after update 5.1

dschaper, 2020-07-16:
I know, it's not PADD that is broken though.

dschaper, 2020-07-16:
`echo 4711 | sudo tee /var/run/pihole-FTL.port` should work temporarily.

Kerdic, 2020-07-17:
v3.2.2 for anyone interested
[padd.txt](https://github.com/pi-hole/PADD/files/4940379/padd.txt)

dschaper, 2020-07-17:
Do you want to PR your changes to see if this will work in the main release?

Profesyor, 2020-07-17:
@Kerdic, of course, we're interested!!!!

Thank you.

Kerdic, 2020-07-17:
> Do you want to PR your changes to see if this will work in the main release?

I will try and look into that this weekend. I would think we should have a more official size name (I was just using PADDcust and custom), and I need to verify the screen size check logic that determines to use the "custom" size, to make sure it wouldn't have ill effects on other screen sizes that should remain using mini,

But yeah, it would be great to get support for this screen into production. I am open to any formatting changes as needed to make the output look cleaner. I know the FTL section isn't up to par (no ===== divider), but I wanted to get that information on the screen. :)

Profesyor, 2020-07-17:
PADD2.8 
in color.

Profesyor, 2020-07-18:
I'm still getting:
Blocking: 0 Domains
Piholed: [ _______________________ ] 0%
Piholed: 0 out of 0
Latest: 0
Top Ad:

Kerdic, 2020-07-18:
Yeah, that is due to a bug in FTL from what I understand.

You will have to get to a command line and run the following below, then it should show the information correctly again. They said it should be fixed in a couple days.

echo 4711 | sudo tee /var/run/pihole-FTL.port

dschaper, 2020-08-30:
"Tiny" is in the master branch. Version 3.3.

closed 2020-08-30 by dschaper

Profesyor, 2020-09-02:
Pi-hole stopped working after these updates 3.3 no internet! I'm DONE with this! tried 3.3.1 same no internet.
Update pi-hole fails. Screen and all in the trash bin! I don't care anymore!

dschaper, 2020-09-02:
PADD has nothing to do with Pi-hole's functions. 

But go ahead and toss the baby out the pram, it's only hurting you.

Profesyor, 2020-09-02:
> PADD has nothing to do with Pi-hole's functions.
> 
> But go ahead and toss the baby out the pram, it's only hurting you.

Not blaming anyone. It was working ok before, I did nothing but update to 3.3 But I'm the reason this thread even exists. 

**Profesyor opened this issue on Mar 4**

dschaper, 2020-09-02:
Again, there is absolutely nothing in PADD that can do anything to stop Pi-hole.

dschaper, 2020-09-02:
> It should have been left alone after 3.2.2

That version doesn't have the tiny display. But feel free to install v3.2.2 and go back to what ever you think works.

Profesyor, 2020-09-02:
So Pi-hole just stops working then?

dschaper, 2020-09-02:
> So Pi-hole just stops working then?

Apparently so. I'm done with this discussion now.

Profesyor, 2020-09-02:
> > So Pi-hole just stops working then?
> 
> Apparently so. I'm done with this discussion now.

And you're a JERK!!!!!!!

mctehlers, 2020-09-20:
So pardon me, but what are the current steps to get this working?

dschaper, 2020-09-20:
Install the latest release according to the README. Everything has been merged.