pi-hole/PADD #87: v3.1 running, yet says Updates are available ?

fireheadman, 2020-03-04:
Unsure if PADD is not recognizing the latest version is installed?
I just updated pihole (sudo pihole -up) and padd.sh

[image]

fireheadman, 2020-03-04:
Solution:
sudo vi padd.sh

CHANGE "Location" to "location"
[image]

[image]

...and now:
[image]

danozdotnet, 2020-03-05:
https://github.com/jpmck/PADD/issues/73

mikegee81, 2020-03-06:
The workaround in #73 did not work for me.
But, this is fixed by the change in pull request in #82

peatrick, 2020-03-07:
Kept getting frustrated, not understanding what I was doing wrong, despite multiple updates -- many thanks, @fireheadman your simple change of "Location" to "**location**" worked perfectly: 
[image]
 and now I don't have to keep banging my head against the wall needlessly.  Only difference, I'm a plebe and use Nano, instead of VIM/VI.  Don't hate me.

p1r473, 2020-03-08:
I am getting the same issue

peatrick, 2020-03-08:
> I am getting the same issue

Did you have any issues making the change above?  Pretty straightforward, just adjust the case sensitivity -- hope that helps resolve your issues, as well.  Fixed it on 3 different Piholes for me, across both baremetal RPi, VMware Debian VM & VirtualBox Ubuntu Server VM.

damienmci, 2020-03-11:
Had the same issue. Made the change and all fixed.

snarkbe, 2020-04-12:
I put a "`grep -i`" to solve the issue. 

Now sure when "location" or "Location" is returned, but the "`-i`" (case insensitive) will cover for both.

fireheadman, 2020-07-10:
UPDATE:
I'm now on the latest version (again)... and back to this issue
[image]

3.2.1 is the latest out, but padd.sh thinks otherwise.
I see that Location is now location, so something else is causing this?
[image]

fireheadman, 2020-07-10:
Looks like something with how github changed urls?
From the padd.sh (version check block)
```
# PADD version information...
    padd_version_latest=$(curl -sI https://github.com/jpmck/PADD/releases/latest | awk -F / 'tolower($0) ~ /^location:/ {print $NF; exit}' | tr -d '\r\n[:alpha:]')
```

So running this.... shows the 301 Moved Permanently
```
pi@pihole01:/etc/pihole/scripts$ curl -sI https://github.com/jpmck/PADD/releases/latest
HTTP/1.1 301 Moved Permanently
server: GitHub.com
date: Fri, 10 Jul 2020 18:41:57 GMT
content-type: text/html; charset=utf-8
status: 301 Moved Permanently
vary: X-PJAX, Accept-Encoding, Accept, X-Requested-With
location: https://github.com/pi-hole/PADD/releases/latest
cache-control: no-cache
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 1; mode=block
referrer-policy: origin-when-cross-origin, strict-origin-when-cross-origin
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; connect-src 'self' uploads.github.com www.githubstatus.com collector.githubapp.com api.github.com www.google-analytics.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events wss://live.github.com wss://alive.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com; frame-ancestors 'none'; frame-src render.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com *.githubusercontent.com; manifest-src 'self'; media-src 'none'; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/socket-worker.js gist.github.com/socket-worker.js
Set-Cookie: _gh_sess=DaUPuNyBhMI3MUrKDUC4n2ibv1fLE2FKdAUCiCllkcYm7sD4WLj%2FXIe3zqkdkzOpMN0nLyRgY25yxgK71AsegfyBrD0D2Me0G3wpy9PzFFSlZwGkil6Jwe%2BBnW21bir2nJseVAq1Nlx6NmWSZ4Zk9LEIujiuYCfz%2F3GuVAj4%2Bqp1IcX8eB%2BKA0%2FnhqIBqp1eKa57FkrC6%2FhsHAu9mAlH9YJ5c%2FEemaUcxzl32g8Wgn9u8JLclriKJiBmvwEkvE%2BBTbvya3mHPBHMxufzLv2pDQ%3D%3D--cJga1ecKAOUl3eyK--xjmnrv%2FLkOTMNYCMKR5OJw%3D%3D; Path=/; HttpOnly; Secure; SameSite=Lax
Set-Cookie: _octo=GH1.1.695771364.1594406517; Path=/; Domain=github.com; Expires=Sat, 10 Jul 2021 18:41:57 GMT; Secure; SameSite=Lax
Set-Cookie: logged_in=no; Path=/; Domain=github.com; Expires=Sat, 10 Jul 2021 18:41:57 GMT; HttpOnly; Secure; SameSite=Lax
Content-Length: 113
X-GitHub-Request-Id: EDD9:4551:26D66:4B978:5F08B675
```

snarkbe, 2020-07-10:
I have opened issue #110 about this issue and how to solve it.

PR #111 (2020-07-12): Use GitHub API for remote version detection.
Use the GitHub API for remote version detection. 

Bumps to version v3.2.2.

Closes #73, closes #82, closes #90, closes #107 
Resolves #110, resolves #97, resolves #73, resolves #80, resolves #87
closed 2020-07-12 by dschaper