pi-hole/PADD #89: PADDmini - update and size?

Sisko4, 2020-03-12:
Hello!
I just installed pihole and the padd plugin, but i have two problems:

1: The Plugin says "PADDmini [i] Update avail."
But i already did an update and reboot everything.
How can i get that solved?

2: Can i use other sizes like the regular info screen on a 2,8" PiTFT?

[image]

Kerdic, 2020-03-12:
[check out this link](https://github.com/jpmck/PADD/issues/86), I have a customized script to support 2.8 screens. But at the moment, there is no official support.

Ignore the update avail until the next official build, it is a bug mentioned in other issues here, and there are multiple pull requests that address the issue. I modified my script to address the bug, but not in the most elegant way.

Sisko4, 2020-03-12:
Ah okay thanks! Yours looks way better than this, they should implemend it!

Profesyor, 2020-03-12:
Also update Pi-hole:
 pihole -up

Sisko4, 2020-03-13:
Already did that before posting, everything up-to-date.

Profesyor, 2020-03-13:
Have you updated line 439 on the original padd.sh from Location to location?

Profesyor, 2020-03-13:
Use @Kerdic latest file, he got that also covered.

dschaper, 2020-08-30:
"Tiny" is now in the master branch, release v3.3

closed 2020-08-30 by dschaper