pi-hole/PADD #91: CPU temp reporting as 0C in PADD, but properly in Pi-hole admin

ilovecats516, 2020-03-16:
*Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!*

**Describe the bug**
The CPU temp reported in PADD is incorrect, at 0C

**To Reproduce**
1. Run PADD
2. Note CPU temp

**Expected behavior**
Correct CPU temp appears, the same temp as the Pi-hole admin interface

[image]

dschaper, 2020-07-12:
What kind of device is Pi-hole running on

zeus86, 2020-08-05:
this problem is caused by the fact, that padd reads temperature from `/sys/class/thermal/thermal_zone0/temp`, while chronometer and FTL try to read this also, but fallback to `/sys/class/hwmon/hwmon0/temp1_input`, which is more likely to be the correct temperature. 

because of the fact, that temperature reading is very different on many systems, it should be user-configurable, where the temp is beeing read from (like the "cputemp/gputemp" approach which kodi uses). I have this issue on every corner with pihole, because my temperature reading on an arm-box with turris (openwrt derivate) has it's temps at `/sys/class/hwmon/hwmon9/temp1_input`.

A related issue to this is this one: 
https://github.com/pi-hole/AdminLTE/issues/1521

dschaper, 2020-08-17:
Putting in code that mimics what chronometer.sh does is not much work to do. Adding in the ability to read user configs means adding a PADD configuration file and a ton of work to read and configure. It would no longer be a single script deploy. I don't see that happening.

zeus86, 2020-08-17:
i don't really see, why padd should not pick up variables from `/etc/pihole/setupVars.conf`. At least to override defaults like "which temperature sensor to read. This way configuration can be uniform across all components of pihole, wether they are considered as "official" or "3rd-party" or "optional". THE one user config is already there in my opinion. `/etc/pihole/setupVars.conf`may also contain an include statement to e.G. `/etc/pihole/conf.d/*.conf` for easier deployment like almost every other unix software nowadays does too.

dschaper, 2020-08-17:
Okay, then that's a feature request.

zeus86, 2020-08-18:
> Okay, then that's a feature request.

this may be partially true, but it would - beside others - fix this bug here, that the temperature is not shown correctly.

dschaper, 2020-08-18:
Be my guest, you'll probably get to it before I would.

zeus86, 2020-08-18:
> Be my guest, you'll probably get to it before I would.

I'd assume you won't like my code if i'd write some, because I'm bad at it :) . In my dayjob I am an admin, and I tend to find the more curious bugs and cornercases, but beside some basics I'm rather bad in coding^^.

dschaper, 2020-08-18:
Try it, I've never seen your code before. If you think your code is bad then take this as a quick opportunity to improve.

thomasmerz, 2022-02-19:
@aidenmitchell , @zeus86 , might this issue also be fixed by PR #146 ?  
(I'm also bad in coding due to my admin-job instead of a developer-job, but I am not too good for anything 🤣 I really wonder and appreciate the patience that yubiuser from Pi-hole team has and he didn't desperate with my PRs)

yubiuser, 2022-02-19:
It's likely fixed by https://github.com/pi-hole/PADD/pull/146.

I link those, if the PR is merged, this will autoclose this issue.

aidenmitchell, 2022-03-11:
I think this was fixed with #198, closing

closed 2022-03-11 by aidenmitchell