pi-hole/PADD #93: Glitch while using pi-hole Beta

WildHunterz, 2020-04-04:
Hi there,

the Display glitches out the range when using the Beta of PiHole caused by the version-numbers.
Any workarround available?

[image]
[image]

jpmck, 2020-04-04:
> _Before submitting a bug report, are you using a test version of Pi-hole? If so, PADD doesn‘t support beta versions of Pi-hole!_

You’re free to modify the code on your install, but there are no plans to have PADD support beta versions of Pi-hole.

closed 2020-04-04 by jpmck

WildHunterz, 2020-04-04:
Thanks for your reply.
Could u give me a hint where to modify the padd.sh?

jpmck, 2020-04-04:
The `PrintLogo()` function might be the easiest place to look. (I’m on my phone so I don’t have the line number handy.)