pi-hole/PADD #95: Extra line in DHCP section

Morrolan, 2020-04-07:
Pi-Hole 4.4, PADD v3.1

**Describe the bug**
The top line of the logo is being cut off due to the DHCP > Router section showing my IP address twice.

Line 331:  DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { print $3 }').  When I run "/sbin/ip route | awk '/default/ { print $3 }'" I actually get 2 results:

10.10.42.1
10.10.42.1

If I run:
```bash
/sbin/ip route
```

I get:

default via 10.10.42.1 dev eth0 src 10.10.42.11 metric 202
default via 10.10.42.1 dev wlan0 proto dhcp src 10.10.42.26 metric 303
10.10.42.0/24 dev eth0 proto dhcp scope link src 10.10.42.11 metric 202
10.10.42.0/24 dev wlan0 proto dhcp scope link src 10.10.42.26 metric 303

So it appears that if wlan0 is present, line 331 detects 2 values (1 for eth0 and 1 for wlan0) and returns both.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable wifi on the Pi, as well as Ethernet
**Expected behavior**
Using the current method on line 331 it is difficult to return a unique value of 2 both eth0 and lan0 are present.  If both eth0 and wlan0 are active and the 2 values returned are the same, only 1 instance should be shown.

**Screenshots**
[image]

Jarsky, 2020-04-17:
I changed mine to handle multiple router addresses on the same line as I have a sub interface on my Pi for my Guest Network so I dont have to allow interVLAN traffic. 

Line 331 I changed this:
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { print $3 }')`

to this:
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { printf("%s\t", $3); next } END {print " "}')`

Screenshot
[image]

Jarsky, 2020-04-17:
I changed mine to handle multiple router addresses on the same line as I have a sub interface on my Pi for my Guest Network so I dont have to allow interVLAN traffic.

Line 331 I changed this:
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { print $3 }')`

to this:
`DHCP_ROUTER=$(/sbin/ip route | awk '/default/ { printf "%s\t",$3 }')`

Screenshot
[image]